GaryOderNichts/Bloopair #1: Manual pair to DS3

Invictaz, 2021-09-19:
If Bloopair can display the WiiU's bluetooth Mac Address we can manually enter that address into the DS3 via a tool called SixaxisPairTool so the controller can sync.

I know that the parent project Bloopair was forked from has some DS3 problems but there are a lot of tools and documentatie already available to fix this.

GaryOderNichts, 2021-09-19:
Dualshock 3 support would be neat. I might look into that.
There are a lot of other things I want to look into first though.

> the parent project Bloopair was forked from
What project do you mean? Bloopair is not a fork.

Invictaz, 2021-09-19:
@GaryOderNichts I mean https://github.com/ndeadly/MissionControl

It looks to me as if Bloopair uses some code from MissionControl. The homebrews seem related. Use other brand controllers on your Nintendo gaming console.

GaryOderNichts, 2021-09-19:
They have the same purpose, but MissionControl runs on a completely different console.
There is no code from MissionControl used in Bloopair.

Invictaz, 2021-09-20:
If we use a modified version of Bluubomb https://github.com/GaryOderNichts/bluubomb we can figure out the bluetooth address of the WiiU. However, this requires a PC. It would be nice if the bluetooth address was displayed on screen, without a pc. You might think that we still need a pc to change the BT-address on the controller but there are Android apps that can do that, such as: https://www.happymod.com/ps3-controller-pairing-tool-mod/com.poke64738.ps3pairingtool/ (not tested)

Invictaz, 2021-09-20:
Well I managed to run a Ubuntu live-dvd  and Bluubomb my WiiU to get it's bluetooth address
So far so good. 

Then set the address with SixAxisPairtool (sigh, another 64bit only tool which installs a driver on Windows) and tried to connect.

[image]

What happens is the DS3 is flashing a lot, but then turns off.

GaryOderNichts, 2021-09-21:
A better solution than displaying the Bluetooth device address and manually pairing the Controller on another device, would be to create a homebrew app that connects to the PS3 controller and writes the pairing directly.
The Wii U supports HID over USB, which makes this a lot easier. See: https://github.com/devkitPro/wut/blob/master/include/nsyshid/hid.h
We then just need to send an HID report with the Wii U's bluetooth address:
https://github.com/user-none/sixaxispairer/blob/master/main.c#L79
The only issue I see is how we get the Wii U's bdaddr from the ARM to the PPC side. Maybe Bloopair could open a IPC node where we can read the address from, or hook into the existing btrm node.

Invictaz, 2021-09-21:
The Bluubomb route where you retrieve the bluetooth address first, then write it with an external app is indeed quite a hassle.
I am not really a fan of an extra app just to pair a controller though, makes it confusing for the end user.
Can you explain why the bdaddr is from ARM and not PPC? How do you retrieve de bdaddr right now? The controller needs something to sync to so it listens for open connections. Those open connections are provided by the WiiU., so I can think of that the address must be known. I have looked into the bt pairing source code but cannot understand the hooking. 

Maybe a stupid question, but can't you just open Bloopair, it copies the bluetooth address into memory.
Then open the pairing app, you plug in the controller and the address gets written from memory.
After that the payload that Bloopair uses is still in memory
Can imagine that the memory is big enough for this.

GaryOderNichts, 2021-09-21:
> I am not really a fan of an extra app just to pair a controller though, makes it confusing for the end user.

I think it's more complicated to use 2 additional apps on a PC instead of just connecting the Controller via USB to the Wii U and running an app.

> Can you explain why the bdaddr is from ARM and not PPC?

Almost everything Bluetooth related runs on the ARM chip. Even most of Bloopair runs on the ARM chip.

> How do you retrieve de bdaddr right now?

Since Bloopair runs in the same IOSU module as the Bluetooth stack, I can just read it from memory.

> Maybe a stupid question, but can't you just open Bloopair, it copies the bluetooth address into memory.

The issue is finding a memory location which can be accessed from the PPC and ARM side without being overwritten by anything.
I think the cleaner solution would be hooking into a dev node.

Anyways I was able to get a good deal on a Dualshock 3 so I'll try to add support for it once it arrives.

Invictaz, 2021-09-22:
If you look into my fork you can spot the changes.
They might help you. They cannot be pull requested yet since I didn't test them yet.

Invictaz, 2021-09-23:
More information on the packet structure that the DS3 expects

https://github.com/felis/USB_Host_Shield_2.0/wiki/PS3-Information