GaryOderNichts/Bloopair #2: Dualshock4 v1 or v2

Invictaz, 2021-09-19:
Not clarified is if Dualshock 4 v1 or v2 is supported.

GaryOderNichts, 2021-09-19:
No Dualshock 4 is supported at the moment.
The only Sony controller supported is the Dualsense (PS5) controller.
DS4 support is planned though, with hopefully both models supported.

Invictaz, 2021-09-21:
Dualshock 4 seems to be already working since it uses the same device name "Wireless Controller"

It might need some tweaks and extra options though from DS4Windows

Invictaz, 2021-09-22:
What about the handshake on DS4? On quite a lot of DS4 related controller emulations a real DS4 should be connected via USB since it enables the 10 min encrypted handshake that is needed to stay connected. For example see using an Xbox controller on a PS4. The DS4 needs to be connected via a hub otherwise after 10 mins the connection drops.

GaryOderNichts, 2021-09-30:
Both should be working with Bloopair version 0.2.0.

closed 2021-09-30 by GaryOderNichts