GaryOderNichts/Bloopair #3: Add option to fix stick drift

Invictaz, 2021-09-19:
Some controllers have stick drift. 

Add option / menu to fix this

Sample homebrew

https://github.com/Xpl0itU/fixyoustick/tree/main

Invictaz, 2021-09-20:
@Xpl0itU So what does your homebrew do then?

Xpl0itU, 2021-09-20:
It works only in the gamepad

Invictaz, 2021-09-20:
Maybe support can be extended to other controllers.

GaryOderNichts, 2021-09-20:
From what I've figured out from Bloopair development the Wii U Pro Controller has a permanent factory calibration. I'm don't think that can be changed though.
The Wii U also reads the normal stick position when switching to a different app or after pairing the controller.
You can see this if you hold the stick in a specific position while pairing it. This is also what happens after you press the button combination @Xpl0itU mentioned.
So a stick calibration tool would be completely useless since the controller calibrates itself every time you connect it.
The only thing that could be adjusted would be the dead zones of the stick.

Invictaz, 2021-09-20:
@GaryOderNichts it recalibrates the WiiU Pro controller. But other controllers are not WiiU Pro controllers, right?

I adjusted the title a bit to better fit the purpose of the request.

GaryOderNichts, 2021-09-20:
Bloopair emulates Wii U Pro Controllers, so if I talk about Pro Controllers here I mean the supported controllers by Bloopair.

luisp316, 2022-02-09:
I have the same issues with a switch pro controller, the left stick doesnt work correctly, it move slowly, for example i cant run in any game, i think is a dead zona issue

onapjd, 2023-02-07:
Have the same issue as @luisp316. My switch pro controller works fine with bloopair except when running left, it goes at a slightly slower speed than the other directions. Connecting my controller to my phone and running a gamepad tester shows that the left direction doesnt reach as high as the other directions. Must be a deadzone issue