GaryOderNichts/Bloopair #7: Build fails

Invictaz, 2021-09-20:
<pre><font color="#4E9A06"><b>ubuntu@ubuntu</b></font>:<font color="#3465A4"><b>~/Bloopair</b></font>$ make
main.cpp
<b>/home/ubuntu/Bloopair/source/main.cpp:41:64:</b> <font color="#CC0000"><b>error: </b></font>&apos;<b>OSDynLoad_NotifyReason</b>&apos; has not been declared
   41 | d load_callback(OSDynLoad_Module module, void *userContext, <font color="#CC0000"><b>OSDynLoad_NotifyReason</b></font> notifyReason, OSDynLoad_NotifyData *infos)
      |                                                             <font color="#CC0000"><b>^~~~~~~~~~~~~~~~~~~~~~</b></font>

<b>/home/ubuntu/Bloopair/source/main.cpp:</b> In function &apos;<b>void load_callback(OSDynLoad_Module, void*, int, OSDynLoad_NotifyData*)</b>&apos;:
<b>/home/ubuntu/Bloopair/source/main.cpp:44:25:</b> <font color="#CC0000"><b>error: </b></font>&apos;<b>OS_DYNLOAD_NOTIFY_LOADED</b>&apos; was not declared in this scope; did you mean &apos;<b>OS_DYNLOAD_UNLOADED</b>&apos;?
   44 |     if (notifyReason == <font color="#CC0000"><b>OS_DYNLOAD_NOTIFY_LOADED</b></font> &amp;&amp; strcmp(&quot;bin\\ghs\\cafe\\cos\\pads\\padscore\\NDEBUG\\padscore.rpl&quot;, infos-&gt;name) == 0) {
      |                         <font color="#CC0000"><b>^~~~~~~~~~~~~~~~~~~~~~~~</b></font>
      |                         <font color="#4E9A06">OS_DYNLOAD_UNLOADED</font>
<b>/home/ubuntu/Bloopair/source/main.cpp:</b> In function &apos;<b>int main(int, char**)</b>&apos;:
<b>/home/ubuntu/Bloopair/source/main.cpp:61:5:</b> <font color="#CC0000"><b>error: </b></font>&apos;<b>OSDynLoad_AddNotifyCallback</b>&apos; was not declared in this scope; did you mean &apos;<b>OSDynLoad_NotifyData</b>&apos;?
   61 |     <font color="#CC0000"><b>OSDynLoad_AddNotifyCallback</b></font>(load_callback, NULL);
      |     <font color="#CC0000"><b>^~~~~~~~~~~~~~~~~~~~~~~~~~~</b></font>
      |     <font color="#4E9A06">OSDynLoad_NotifyData</font>
make[1]: *** [/opt/devkitpro/devkitPPC/base_rules:13: main.o] Error 1
make: *** [Makefile:100: build] Error 2
</pre>

Invictaz, 2021-09-20:
Any idea what is wrong?

devkitARM
devkitPRO
wut

are all installed

GaryOderNichts, 2021-09-21:
You seem to be using an old version of wut. Try building the wut master branch from source.

Invictaz, 2021-10-03:
Fixed with newer version of wut, with "make install".
And remember to install devkitARM from 3ds-dev

Also check if your code is having the right amount of (( and )) brackets in your code.

closed 2021-10-03 by Invictaz