GaryOderNichts/Bloopair #12: 8Bitdo compatible with BlooPair

aknologias, 2021-09-25:
It will be possible to support an 8Bitdo Sf30 Pro controller with bloopair.

GaryOderNichts, 2021-09-26:
Don't expect any controllers that aren't listed on the README to work. I don't own this controller so I currently won't be able to add support for it.
I'll mark this as a feature request for the future.

kazcio, 2021-09-27:
I happen to have the controller in question along with Linux PC and Wii U capable of running the BlooPair.

If I understand correctly, testing is done via modified version of BlueZ server on linux VM / OS.

I am happy to generate / provide logs if that helps?

GaryOderNichts, 2021-09-30:
Do the 8-Bitdo controllers not work when set to emulate a Xbox / Switch / DS4 controller?
As far as I know they can be set to emulate one of them which should work with Bloopair.
Can anyone try this?

kazcio, 2021-10-01:
Yes, you can run them in different modes. My issue was resolved by updating firmware for my controller. This was a bug on 8bitdo side. 

Documentation
https://download.8bitdo.com/Manual/Controller/SN30-Pro+/SN30_Pro+_Manual.pdf

Firmware v5.02
1. Optimized the accuracy of joystick.
2. Fixed the vibration issue on Switch mode.

The vibration issue was that when you connected - vibration would never stop after initial connection, it would continue to vibrate. 

As we have 2 reports of the controller working in switch mode with Wii u, it can be marked as working with newest bloopair version.

I am happy to test it further with software of your choice, apologies for the delayed reply!

kazcio, 2021-10-01:
Yes, you can run them in different modes. My issue was resolved by updating firmware for my controller. This was a bug on 8bitdo side. 

Documentation
https://download.8bitdo.com/Manual/Controller/SN30-Pro+/SN30_Pro+_Manual.pdf

Firmware v5.02
1. Optimized the accuracy of joystick.
2. Fixed the vibration issue on Switch mode.

The vibration issue was that when you connected - vibration would never stop after initial connection, it would continue to vibrate. 

As we have 2 reports of the controller working in switch mode with Wii u, it can be marked as working with newest bloopair version.

I am happy to test it further with software of your choice, apologies for the delayed reply! 

To sum it up, set the connetion mode to switch, make sure that you are on the latest firmware for your controller. 5.02 if I remember is the latest.

Thanks for amazing software! I love it! @GaryOderNichts

ssapalski, 2021-10-01:
> Do the 8-Bitdo controllers not work when set to emulate a Xbox / Switch / DS4 controller? As far as I know they can be set to emulate one of them which should work with Bloopair. Can anyone try this?

I just installed Bloopair 0.2.0 and paired a 8Bitdo SF30 Pro (Switch Mode / Start + Y) with my WiiU. I've tested using it with Super Mario 3D World and it works without any issues 👍 This means the SN30 Pro should work as well since to my understanding the only difference between SF30/SN30 is the design but from HW point of view, they are the same.

ssapalski, 2021-10-01:
Unfortunately I just encountered an issue in e.g. Super Mario 3D Word.

If one uses the left analog stick and pushes it to left (walk left), no other button presses are recognized afterwards until the stick is e.g. moved a bit up/down. This results in if walking left in the game, you can't jump anymore until you move the stick a bit up/down. Buttons overall work fine individually.

This problem does not appear using the DPad or using the left analog stick and pushing it to the right (walk right). My SF30 Pro was on firmware version 1.33 and I updated it just now to latest version 1.34 to see if this solves the issue which is unfortunately not the case.

aknologias, 2021-10-01:
I just tested the 8Bitdo SF30 Pro together with BlooPair V0.2.0, I tested its use with Breath of the Wild and the left stick when moving left blocks any other action button (can't jump left ← + B). On top of this, it doesn't recognize motion control, that's all I've tried so far, and otherwise I updated my 8Bitdo SF30 Pro to V1.37 with the "Firmware Updater" tool.
Finally
I am very happy that this 8Bitdo control is compatible with Bloopair, but it would be great to fix this problem (← + A, B, X, Y)

GaryOderNichts, 2021-10-02:
Can you see if this build fixes that issue? Just copy the .rpx inside of the .zip to `wiiu/apps/Bloopair`.

[Bloopair.zip](https://github.com/GaryOderNichts/Bloopair/files/7271660/Bloopair.zip)

kazcio, 2021-10-02:
Hi @GaryOderNichts 

I have uploaded the .rpx file to the  to /wiiu/apps/Bloopair directory.

Tested with Zelda BOTW, I cannot jump to the left. (Left analog stick + X) 

I am on 8bitdo SN30+ Pro controller.

Super Mario 3D world - same issue. I cant jump to the left with the analog stick. D-pad + left is working OK.

kazcio, 2021-10-02:
I have came across another odd behaviour, long press of the start button (which should turn the controller off) scrolls the screen to the right in the main menu.

kazcio, 2021-10-02:
When you close down eshop game (Super Mario World), it still soft locks the console with the music playing in the background. 

Tested on the file you provided + after re-syncing the controller.

ssapalski, 2021-10-02:
> Can you see if this build fixes that issue? Just copy the .rpx inside of the .zip to `wiiu/apps/Bloopair`.
> 
> [Bloopair.zip](https://github.com/GaryOderNichts/Bloopair/files/7271660/Bloopair.zip)

Just tested (controller was re-paired), issue still exists. There are some other issues as well but maybe we could focus first on the issue: **walk left with left analog stick blocks all other action buttons**. If this is solved, the controller would be already useable.

@GaryOderNichts Since you don't have such a controller, is there any way we could provide debug information to you? Is there any recommended WiiU remote debug documentation available and some advices from you how to debug this on our side?

GaryOderNichts, 2021-10-02:
One last try without remote debugging. Does this fix the walk left issue?

[Bloopair.zip](https://github.com/GaryOderNichts/Bloopair/files/7272003/Bloopair.zip)

ssapalski, 2021-10-02:
> One last try without remote debugging. Does this fix the walk left issue?
> 
> [Bloopair.zip](https://github.com/GaryOderNichts/Bloopair/files/7272003/Bloopair.zip)

Yes, this works 👍 Problem fixed, thanks a lot 👍 
Shall we discuss the other minor issues here or open new tickets?

GaryOderNichts, 2021-10-02:
You can just mention other issues here. This is basically about getting the 8BitDo controllers working properly.

You can also reach me on [discord](https://discord.gg/geY4G2NZK9).

ssapalski, 2021-10-02:
Other issues are:

1. After starting Bloopair, "WUP GX2 Installer" fails to start within HBL. Console needs to be restarted in order to make it work. This seems to be a general issue not related to 8Bitdo controllers.
2. After connecting SF30 Pro Controller, it works in HomeScreen and to start a game. After the game (e.g. SM3D World) has been started, it either is disconnected immediately or not responsive. After re-connecting it, it can be used fine within the game.
3. My SF30Pro controller disconnects quite fast if inactive. Haven't measured the time so far but it feels like 1-2 minutes.

GaryOderNichts, 2021-10-03:
I think 1. and 2. are both caused by an issue with the Homebrew Launcher Channel described here: https://github.com/GaryOderNichts/Bloopair/issues/17#issuecomment-932974462
I can't really do anything about that.

About 3: It's probably the controller disconnecting, since the Wii U doesn't send any reports to it when it's not used.
Maybe there is some sort of heartbeat report that's sent by an original switch.

aknologias, 2021-10-03:
Other problems I have encountered are:

1. If the paired control is not the first control (control number 1), it pairs but does not work.

2. Any other professional controller paired as control number two also stops working, after pairing the 8Bitdo.

Maybe there will be the possibility of emulating the motion control of a Wii Remote Plus like this 
https://cdn.vox-cdn.com/thumbor/deDzKsTaN3jVZJi5JaTWLmpVaFg=/79x0:943x576/1400x1400/filters:focal(79x0:943x576):format(jpeg)/cdn.vox-cdn.com/uploads/chorus_image/image/46434834/tumblr_np5f0sgTcS1u3akyno1_1280.0.0.jpg

RodrigoTomeES, 2022-02-24:
Hi,

I have a SF30 PRO but I can't connect it to the Wii U, has anyone the same issue?

Thanks!

kazcio, 2022-02-25:
For anyone having issues with the 8bitdo controllers, please make sure that you have the latest available firmware from:
https://support.8bitdo.com/

commit 73361f53 (2022-04-05): Change xbox rumble behaviour
Closes #12, closes #41
closed 2022-04-05 by GaryOderNichts