GaryOderNichts/Bloopair #13: Does it work with an Ipega or PowerA controller?

BerserkSoul1989, 2021-09-26:
Hi, thanks for making this homebrew.
As the question says, does this work with generic bluetooth controllers, like Ipega, et?

GaryOderNichts, 2021-09-26:
Don't expect any controllers that aren't listed on the README to work. I don't own any of those controllers so I currently won't be able to add support for them.
I'll mark this as a feature request for the future.

manaboy, 2021-10-17:
How about Xbox 360 controller? I say this because of the Microsoft wireless control attachment for the PC it's given good use for these controller on the pc long after its Xbox 360 RROD.

shanez1215, 2021-11-07:
> How about Xbox 360 controller? I say this because of the Microsoft wireless control attachment for the PC it's given good use for these controller on the pc long after its Xbox 360 RROD.

The protocol for the 360 Wireless controllers is not Bluetooth but a proprietary 2.4GHz signal (closer to wifi) so that'd be out of the scope of this.

closed 2026-09-27 by GaryOderNichts