GaryOderNichts/Bloopair #14: Soporte para control Xbox series x/s

JaiirGarciaBv, 2021-09-26:
Mira conecte mi control de Xbox one y todo bien 
Pero al querer conectar el control de Xbox series s no me empareja, no empareja, pero gracias por tu trabajo es muy bueno la verdad es que los controles de Wii u son caros aquí en México, pero si pudieras poner soporte a series x/s sería lo mejor gracias

GaryOderNichts, 2021-09-26:
The Series X and S controllers are using Bluetooth LE which isn't supported at the moment.
I might be able to support them in the future.

Benzant, 2021-09-27:
I have the same problem but I have in firwareas abasado for being an insider and it left me for 1 minute and mad never worked :/

Benzant, 2021-09-27:
I have heard that HDpad has an app to use the Xbox control for the PC, can you do an ogiañ but for Android?

shanez1215, 2021-11-07:
If you have the Xbox Wireless USB adapter, you may be able get support using HID to VPAD as a workaround.

LCH1238, 2022-06-11:
>  I aslo use Xbox series s controler, thanks for your work！

rusa99ec, 2022-12-26:
> If you have the Xbox Wireless USB adapter, you may be able get support using HID to VPAD as a workaround.

Does that work tho? I tried with mine and doesnt

rusa99ec, 2022-12-26:
I second this! Please make it real. Having a controller that works on PC and Wii U is just a banger and also helps saving money. I bought an Xbox One (3rd gen) controller but I already have two Xbox Series Controllers laying around and I guess I'm not the only one with this kind of setup (PC+Wii U).

MrHaram, 2025-09-25:
Are there still plans to add xbox series controller support?

GaryOderNichts, 2026-10-03:
There are experimental Bloopair test builds available at https://github.com/GaryOderNichts/Bloopair/discussions/217 with support for BLE controllers.

GaryOderNichts, 2026-10-25:
[Version 1.1.0](https://github.com/GaryOderNichts/Bloopair/releases/tag/v1.1.0) now supports BLE controllers.

closed 2026-10-25 by GaryOderNichts