GaryOderNichts/Bloopair #16: Switch SNES Controller Support

NoahOberstein, 2021-09-27:
I believe the Switch SNES Controller (bluetooth) can work with Bloopair. As a proof of concept I edited
this block of code in controllers.c to allow the controller to pair:
```
if ((_strncmp((const char*) name, "Pro Controller", 0x40) == 0) || // switch pro controller
            (_strncmp((const char*) name, "SNES Controller", 0x40) == 0) || // switch snes controller
            (_strncmp((const char*) name, "Joy-Con (R)", 0x40) == 0) || // joycon r
            (_strncmp((const char*) name, "Joy-Con (L)", 0x40) == 0)) { // joycon l
            controllerInit_switch(&controllers[handle], 
                _strncmp((const char*) name, "Pro Controller", 0x40) == 0 || _strncmp((const char*) name, "SNES Controller", 0x40) == 0, // add in snes stuff
                _strncmp((const char*) name, "Joy-Con (R)", 0x40) == 0);
            return 0;
        }
```

and disabled this check in switch_controller.c:

```
//if (!isProController) {
        // The pro controller already sends continous reports but the joy-con don't
        initContinuousReports(controller);
    //}
```

My guess is that the controller is laid out like a Switch Pro Controller missing a few buttons. Forcing the `initContinuousReports` function to run regardless of switch controller type fixed issues with buttons being stuck on the SNES controller. I've tested it on my own Wii U, playing a virtual console game, and everything seems to work.

Obviously this is a bit of a hacky implementation, but it shows it can be done.

Thanks for making this awesome project!

GaryOderNichts, 2021-09-28:
Ah good to know.
The next release will use vendor and product IDs instead of names, so support for the SNES controller should be easily addable.
I've also reworked the reporting process quite a bit.

I will look into supporting it in the next release.

GaryOderNichts, 2021-09-30:
Should be working with Bloopair version 0.2.0.

closed 2021-09-30 by GaryOderNichts