GaryOderNichts/Bloopair #19: [REQ] Clone Switch Controller support

Ploggy, 2021-10-02:
My Clone Switch Controller already works (for the most part) there's only a couple odd issues...
Both Analog Sticks up and down are swapped and the A,B,X,Y buttons are swapped.. all the rest of the Buttons are fine.

Left Analog up should be - Analog down
Left Analog down should be - Analog up 
Right Analog up should be - Analog down
Right Analog Down should be - Analog up 
A should be - B
B should be - Y
X should be - A
Y should be - X

Then the other issue is when Left Analog left is held all other Buttons stop working.. until I release the Stick :P

GaryOderNichts, 2021-10-02:
> Both Analog Sticks up and down are swapped and the A,B,X,Y buttons are swapped.. all the rest of the Buttons are fine.

That might be because Bloopair doesn't use the extended reports of the switch controller. Your clone probably uses a different mapping with basic reports for usage on a PC.

> Then the other issue is when Left Analog left is held all other Buttons stop working.. until I release the Stick :P

That was also an issue with 8BitDo controllers. See if the build attached here fixes that: https://github.com/GaryOderNichts/Bloopair/issues/12#issuecomment-932799491

Ploggy, 2021-10-02:
Sweet, Yeah, it's one of those cheap Clones that us the same shell as a legit Pro Controller, considering it's a Clone it feels really nice in the hand and is only missing HD rumble and NFC, so it would make a good backup Controller I'll look forward to the future release.  :)

Aha, Left issue is fixed with that build :) Thank You!

GaryOderNichts, 2021-11-03:
Finally had time to implement the extended reports. Can you see if everything works with [this build](https://github.com/GaryOderNichts/Bloopair/files/7468722/Bloopair.zip)?

Ploggy, 2021-11-03:
Since this is now working correctly, I'll close the issue :)

closed 2021-11-03 by Ploggy