GaryOderNichts/Bloopair #23: Desconexión de mando de PS4 de forma abrupta

DrakensanhLuger647, 2021-10-14:
La desconexión de este mando puede ser perjudicial en contadas ocasiones pero frecuentes afectando a los otros mandos conectados a la consola obligando a apagar la misma para solucionar el error.

DrakensanhLuger647, 2021-10-14:
Mando testeado con varios juegos en multijugador para ver su funcionalidad y su comportamiento en títulos de todo tipo.

GaryOderNichts, 2021-10-14:
Any chance you could translate that to English?
It's quite hard to understand since I don't speak Spanish.

DrakensanhLuger647, 2021-10-14:
[image]

DrakensanhLuger647, 2021-10-14:
I was playing Tekken and it disconnects but it appears connected with the red light when I went to unlink the control, the original pro control was unlinked and then it did not reconnect

DrakensanhLuger647, 2021-10-14:
I'm going to test as primary control

closed 2024-07-20 by GaryOderNichts