GaryOderNichts/Bloopair #25: [Request] Add Support for PowerA "Enhanced Wireless Controller for Nintendo Switch"

DerpyBubblez, 2021-10-16:
It looks like a Switch Pro Controller, feels like a Switch Pro Controller, works like a Switch Pro Controller, but it is not a Switch Pro Controller.

I spent over $40 when I could've got it for $17. [Please make it work.](https://www.gamestop.com/gaming-accessories/controllers/nintendo-switch/products/powera-enhanced-wireless-controller-for-nintendo-switch-animal-crossing-new-horizons-timmy-and-tommy-nook/11100312.html?editionexpanded=true)

shanez1215, 2021-10-17:
Good evening, I have a Power A Gamecube controller [(This one)](https://www.amazon.com/PowerA-Pokemon-Wireless-GameCube-Controller-Nintendo/dp/B07PP74ZQV) and it turns out Power A set mine up exactly the same as your controller. When running the debug build my vendor and product Id's were _also_ both zero (Maybe ALL of their controllers are just the same schematic in a different shell?)

Thus I was able to grab your new build from this thread and it worked perfectly! All I had to do in Smash for Wii U was remap the Pro Controller to match the gamecube layout.

GC:
A: Attack
B: Special
X: Jump
Y: Jump

Wii U Pro Controller
A: Jump
B: Special
X: Jump
Y: Attack

For other games that don't let you remap, it may be a good idea to have this mapping be swapped by default. Since the SNES Nintendo likes to use the leftmost button as the "main" one in many games with the bottom being one you press with the lower part of your thumb (example running and jumping in Mario) and the top and right being quick presses. On the Wii U and Switch that button ends up being Y instead of A, but the GC's odd layout makes that button A. But then you'd have to use Y to go through the menus which would feel off. Grr.

GaryOderNichts, 2021-11-20:
Reworked Bloopair to check for Switch controller names before using the VID / PID.
Would be cool if someone could test this build with a Power A controller and report back if it works:
[Bloopair.zip](https://github.com/GaryOderNichts/Bloopair/files/7573665/Bloopair.zip)

I just hope there aren't any other controllers named "Pro Controller" or else this might cause issues.

DerpyBubblez, 2021-11-20:
I've just tested this build, and it seems to work fine. I just had to re-pair my controller.

GaryOderNichts, 2021-11-24:
These changes were implemented in version 0.5.0

closed 2021-11-24 by GaryOderNichts