GaryOderNichts/Bloopair #33: Map (parts of) a controller to the GamePad

EnterpriseFreak-v2, 2021-11-10:
Some games such as Zelda Breath of the Wild require motion controls in some areas so it would be nice to map controllers that have a built-in gyro sensor (partially) to the GamePad so to use that feature.

GaryOderNichts, 2021-11-11:
This is not possible, since Bloopair currently does not emulate a GamePad.
I'm not sure how this would work, since Games only expect one GamePad to be connected.
Maybe if Bloopair ever implements a config system, one controller could be set to replace the gamepad inputs.

Sonicjan, 2024-05-29:
Instead of emulating a Gamepad, wouldn't it be possible to force the System to accept Gyro from Pro Controllers?

iam-best, 2025-08-13:
Is it possible to emulate a Wii remote instead with gyro? Would be useful for joycons for example.

GaryOderNichts, 2026-10-25:
GamePad emulation is not planned at the moment.

closed 2026-10-25 by GaryOderNichts