GaryOderNichts/Bloopair #38: Wii U freeze

BaamAlex, 2021-12-18:
The program works perfectly. But sometimes the console freezes when i leave the system settings or close a game. It stucks in the "Wii U Menu" screen forever. Just holding the power button can shut down the console.

GaryOderNichts, 2021-12-19:
Are you using the Homebrew Launcher Channel to launch Bloopair?
There seems to be an issue with the Homebrew Launcher Channel. This isn't something that can be fixed in Bloopair itself.
See the following [issue comment](https://github.com/GaryOderNichts/Bloopair/issues/17#issuecomment-932974462) for more info.

BaamAlex, 2021-12-19:
I use a channel from a user from GBATemp.

GaryOderNichts, 2021-12-21:
That channel might contain the same bug the Homebrew Launcher Channel does. Does launching Bloopair from the normal Homebrew Launcher have the same freezes?

GaryOderNichts, 2022-01-08:
This should be no longer relevant with the Tiramisu version.

closed 2022-01-08 by GaryOderNichts