GaryOderNichts/Bloopair #40: Agregar más controladores de PS3 Dualshock

Missael03, 2021-12-25:
Tengo un control de PS3 y no se conecta porque no tiene un ID válido

GaryOderNichts, 2021-12-25:
Please keep it in English here, as I don't speak your language.

What exactly doesn't work? Does the DS3 get detected when you plug it in on the pairing menu?

Missael03, 2021-12-25:
I have a clone and the ID is not valid it says 000.000.000

GaryOderNichts, 2021-12-25:
What ID exactly? Can you upload a screenshot of the pairing screen?

Missael03, 2021-12-25:
[image]

GaryOderNichts, 2022-01-08:
Looks like this clone doesn't implement the report Bloopair uses to read the address of the controller.
There isn't really something I can do, without having an alternative way to read the address.

GaryOderNichts, 2022-01-18:
Please test if version 0.5.1 fixes this issue.

Missael03, 2022-01-22:
still not connecting

czerwinski76, 2022-12-24:
Same thing with my DS3 controller.