GaryOderNichts/Bloopair #41: 8BitDo Pro 2 Support

Redcarred2, 2021-12-28:
Hello, The WiiU cant detect the 8BitDo Pro 2 after initializing Bloopair.

chrislachance, 2021-12-31:
Original 8BitDo Pro works just fine. I paired as an XBox controller

GaryOderNichts, 2022-01-08:
I think this controller uses Bluetooth Low Energy, which is currently not supported.
Looking into adding support for it though.

lm7nyne, 2022-01-26:
I was able to get my Pro 2 working by moving the switch on the back to X and then syncing. Only problem I've noticed is that the rumble motors won't stop once connected.

Redcarred2, 2022-02-19:
> I was able to get my Pro 2 working by moving the switch on the back to X and then syncing. Only problem I've noticed is that the rumble motors won't stop once you move the hand over one of the icons on the screen. I saw another bug post for sn30 that had the same issue but that was fixed with a firmware update. My Pro 2 is up to date so I am at a loss.

This is a fix, I just made a profile with Rumble set to zero in the pro software

commit 73361f53 (2022-04-05): Change xbox rumble behaviour
Closes #12, closes #41
closed 2022-04-05 by GaryOderNichts