GaryOderNichts/Bloopair #42: Gyroscope support?

chrislachance, 2021-12-31:
I think you probably are aware, but gyroscope support would be great in the future if not already supported.

dude98, 2022-07-17:
The Wii U pro controller doesn't have a Gyroscope, so unless they eventually start emulating a gamepad or wiimote plus (or wii motion plus), there's no point in adding a gyroscope.

monkeyman4412, 2022-08-28:
There is a use for this, albeit, one tiny microscopic case. Splatoon's local play. It would be hell of a lot less janky, than what the game asks you to do. (which is to essentially ruberband a wiimotion plus controller to the bottom of your full controller, to play locally)

monkeyman4412, 2022-08-28:
There is, unironically, a use for this. Setup the gyro as a wiimotion plus controller ontop of the pro controller. And now you can use that configuration for splatoon's local multiplayer

chrislachance, 2022-08-28:
This is the exact reason I'm hoping for it. My kids love that mode.

Sentineluno, 2023-04-22:
Me too, was just playing breath of the wild but it's not comfortable to play without gyro aim for the bow aiming and even certain shrines, it may not be possible but if there is i would love to see this being a thing.

anegrin, 2023-10-22:
>  And i mean when using for example a switch pro controller.

yeah this would be great as it would mean you can also use a 8bitdo pro 2 and gyro... +1 for the feature

anegrin, 2023-11-04:
oh here's why this can't happen

https://github.com/GaryOderNichts/Bloopair/issues/33#issuecomment-966099002

chrislachance, 2024-08-17:
That makes sense, but I believe (a least for Splatoon) it takes the input of TWO controllers; gyro of one, inputs of the other.
Wonder if that's possible.

(How it worked - https://www.youtube.com/watch?v=vBlNDDZRf-4)

closed 2026-09-30 by GaryOderNichts