GaryOderNichts/Bloopair #44: some apps froze on boot esp. on tiramisu

tehrzky, 2022-01-04:
-hyrule warriors
test 1 with these module installed
module:
bloopair
hbl_installer 

when booting its stuck on boot

test 2 without hbl_installer
module:
bloopair 

it will run but if you have DLC it will also stuck on boot

test 3 without module hbl_installer & bloopair

it will run perfectly with DLC, to use bloopair you need to run this inside homebrew loader then run hyrule warriors.

after that if you have a save game it will not stuck anymore. (not sure, but i tried to install the module bloopair it boot with dlc)

-youtube app
same with Hyrule warriors, but if you have bloopair and you manage to boot youtube then exit the app it will stuck on exit.

maybe this is the same case for other games that froze on boot.

tnx.

GaryOderNichts, 2022-01-05:
I don't think this is related to Bloopair.
Please download the latest version of the hbl_installer. The hbl_installer was updated a few minutes ago to fix crashes in some games (https://github.com/wiiu-env/HBLInstallerWrapper/commit/442d1d3a274d840b1ac0ee57d3d9dd1d5929d326).
Also make sure you're using the Tiramisu Edition of Bloopair, and don't launch the old version through the Homebrew Launcher.

tehrzky, 2022-01-06:
idk where to put this issue, sorry. because when i put bloopair tiramisu version it cannot boot the game with DLC (test 2 & 3) so i decided to post here.  tnx for the hard work

Cqtuanptit, 2022-01-10:
i have same problem with mass effect 3

GaryOderNichts, 2022-01-10:
Please try deleting the `wiiu/environments/tiramisu/modules/setup/50_hbl_installer.rpx` file.
If that fixes your issue, please report this issue here instead: https://github.com/wiiu-env/HBLInstallerWrapper

SykkNyzz, 2022-01-27:
Wait... Tiramisu was updated when?