GaryOderNichts/Bloopair #45: PS5 Controller Battery Icon

BaamAlex, 2022-01-06:
I use a ps5 controller atm. And the battery level in the wii u pause menu (the one when you press the home button in-game) is weird. It shows me charging, fully charged, half charged or almost empty. And this changes as good as every 1 or 2 seconds.

Dunno if my description is good. But I hope you know what i mean. The controller itself works perfectly fine.

GaryOderNichts, 2022-01-07:
Seems like the battery levels for the DS in Bloopair are off by one in the input report.
Can you test if this build works? (Copy the `30_bloopair.rpx` to `wiiu/environments/tiramisu/modules/setup`)
[30_bloopair.zip](https://github.com/GaryOderNichts/Bloopair/files/7831777/30_bloopair.zip)

BaamAlex, 2022-01-08:
I tested it now and this happens.

https://drive.google.com/file/d/13cU6hZBrHPPZhslwSXoYU3isoUPeaYc-/view?usp=drivesdk

GaryOderNichts, 2022-01-08:
What about this one?
[30_bloopair.zip](https://github.com/GaryOderNichts/Bloopair/files/7834121/30_bloopair.zip)

Sorry, I don't have a dualsense at the moment, so I can't do any testing myself.

BaamAlex, 2022-01-09:
Sadly no.

GaryOderNichts, 2022-01-09:
Hmm, I have no idea what could cause this.
Is this an official DualSense?

BaamAlex, 2022-01-09:
Definitely yes.

BaamAlex, 2022-01-09:
If you can't fix it now, don't worry. It is nothing what bothers me. I just noticed it and wanted to report that bug. That's all.

GaryOderNichts, 2022-01-18:
Fixed with version 0.5.1, thanks for reporting.

closed 2022-01-18 by GaryOderNichts