GaryOderNichts/Bloopair #47: Want to add my own controllers

chr1s-t0pher, 2022-01-08:
which files will i need to modify in order to add another gamepad?
like nvidia shield gamepad vendor_id = 0x0955 and product_id = 0x7214
or my amazon firetv gamepad vendor_id = maybe 1949 and product_id = maybe 0402

would copy probably the xbox_one_controller.c and rename all to shield_controller, but which other files do i need to modify?
and than i would test it with Controller Test by brienj

I can build files i setup devkitpro.

more familuar with c sharp and python.

GaryOderNichts, 2022-01-08:
You'll have to create a new .c file for a controller in `source/ios_pad/source/controllers`. You can base this file on one of the existing controllers.
In the `controllerData` function you'll need to parse the incoming input reports and set the values in `reportData` accordingly. 
Stick values should be normalized between -1140 and +1140.

Then you can add the `controllerInit` function declaration after the other declarations here: https://github.com/GaryOderNichts/Bloopair/blob/9ce7b47be0d5dd1f597b406e97c28d15bee56d87/source/ios_pad/source/controllers.c#L113
and add a branch for your vendor and product IDs like this one here:
https://github.com/GaryOderNichts/Bloopair/blob/9ce7b47be0d5dd1f597b406e97c28d15bee56d87/source/ios_pad/source/controllers.c#L186

chr1s-t0pher, 2022-01-10:
Amazon FIreTV Gamepad Model no. WR26UR vendor_id=1949 product_id=0402

static const uint32_t dpad_map[9] = {
    0,
    WPAD_PRO_BUTTON_UP,
    WPAD_PRO_BUTTON_UP | WPAD_PRO_BUTTON_RIGHT,
    WPAD_PRO_BUTTON_RIGHT,
    WPAD_PRO_BUTTON_RIGHT | WPAD_PRO_BUTTON_DOWN,
    WPAD_PRO_BUTTON_DOWN,
    WPAD_PRO_BUTTON_DOWN | WPAD_PRO_BUTTON_LEFT,
    WPAD_PRO_BUTTON_LEFT,
    WPAD_PRO_BUTTON_LEFT | WPAD_PRO_BUTTON_UP
};

#define AXIS_NORMALIZE_VALUE (1140 * 2)

void controllerData_amazon(Controller_t* controller, uint8_t* buf, uint16_t len)
{
    ReportBuffer_t* rep = controller->reportData;

    if (buf[0] == 0x01) {
        rep->left_stick_x = (buf[1] - 256 / 2) * AXIS_NORMALIZE_VALUE / 256;
        rep->left_stick_y = (buf[2] - 256 / 2) * AXIS_NORMALIZE_VALUE / 256;
        rep->right_stick_x = (buf[3] - 256 / 2) * AXIS_NORMALIZE_VALUE / 256;
        rep->right_stick_y = (buf[4] - 256 / 2) * AXIS_NORMALIZE_VALUE / 256;

        // clear all buttons besides home
        rep->buttons &= WPAD_PRO_BUTTON_HOME;

        if ((buf[9] & 0xf) < 9)
            rep->buttons |= dpad_map[buf[9] & 0xf];
        if (buf[5])
            rep->buttons |= WPAD_PRO_TRIGGER_ZL;
        if (buf[6])
            rep->buttons |= WPAD_PRO_TRIGGER_ZR;
        if (buf[7] & 0x01)
             rep->buttons |= WPAD_PRO_BUTTON_B;
        if (buf[7] & 0x02)
             rep->buttons |= WPAD_PRO_BUTTON_A;
        if (buf[7] & 0x08)
             rep->buttons |= WPAD_PRO_BUTTON_Y;
        if (buf[7] & 0x10)
             rep->buttons |= WPAD_PRO_BUTTON_X;
        if (buf[7] & 0x40)
             rep->buttons |= WPAD_PRO_TRIGGER_L;
        if (buf[7] & 0x80)
             rep->buttons |= WPAD_PRO_TRIGGER_R;
        if (buf[8] & 0x08)
             rep->buttons |= WPAD_PRO_BUTTON_PLUS;
        if (buf[8] & 0x20)
             rep->buttons |= WPAD_PRO_BUTTON_STICK_L;
        if (buf[8] & 0x40)
             rep->buttons |= WPAD_PRO_BUTTON_STICK_R;
        if (buf[8] & 0x04)
             rep->buttons |= WPAD_PRO_BUTTON_MINUS;
        
    }
    else if (buf[0] == 0x02) {
        if (buf[1] & 0x10)
            rep->buttons |= WPAD_PRO_BUTTON_HOME;
        else
            rep->buttons &= ~WPAD_PRO_BUTTON_HOME;
    }
    else if (buf[0] == 0x04) {
        controller->battery = (buf[1] & 0x3) + 1;
        controller->isCharging = buf[1] & 0x10;
    }
}

GaryOderNichts, 2022-01-21:
Sorry for the late reply.
Did you test the code and it's fully working?
Are there any references for the reports of this controller? Seems a bit odd that it also uses report 2 and 4 for home button presses and battery levels.
If everything works, could you open a PR for it?

chr1s-t0pher, 2022-01-21:
Battery level not tested, used "Hid Demo" from SharpLibHid to figure the hex codes out. Buttons, direction buttons and analog sticks work, all tested with a tool for the wii u, also played a couple of games.

Bis jetzt funtioniert alles, aber weiss noch nicht wie ich die lampen ansteuere und rausbekomme wie voll die Batterie ist.

closed 2025-04-07 by GaryOderNichts