GaryOderNichts/Bloopair #50: No puedo apagar la consola desde el gamepad cuando activo bloopair, es normal?

loger0509, 2022-01-12:
Buenas tardes, tengo poco desde que  uso bloopair v0.5.0, actualmente uso Haxchi  y activo el bloopair desde la hombrew y vinculo un control de switch y se juega perfecto, el problema se da cuando quiero apagar la consola desde el gamepad, presiono el botón de Power del gamepad y se apaga, pero la consola se queda prendida por mucho tiempo y tengo que apagarla desde el botón de Power de la misma, es normal este comportamiento?, estoy pensando en instalar tiramisu para tener el bloopair activado desde que arranca la consola, pero me surge la misma duda que me pasa con haxchi sobre el apagado de la consola, espero me puedan orientar, muchas gracias.

Saludos desde Sinaloa, Mexico...😁

GaryOderNichts, 2022-01-13:
Please keep it in English here, as I don't speak your language.

Are you using Tiramisu? Because this error only occurs when using Bloopair through the Homebrew Launcher Channel.

loger0509, 2022-01-13:
good morning, I haven't switched to tiramisu yet, I currently use Haxchi and run bloopair from Hombrew Launcher.

GaryOderNichts, 2022-01-18:
This is an issue with the Homebrew Launcher Channel, please switch to Tiramisu or use the normal Homebrew Launcher instead.

closed 2022-01-18 by GaryOderNichts