GaryOderNichts/Bloopair #51: Mario Kart 8 not working with Bloopair Tiramisu

roxas10, 2022-01-16:
Hi, so I installed Tiramisu on my wii u with the bloopair module. I connected my switch controller (Its a OEM from aliexpress), and all the games work great, except Mario Kart 8 so that it won't work in the game, but on home menu yes. That also happens with the N64 Virtual Console and Super Mario 64.

GaryOderNichts, 2022-01-16:
What do you mean by "not working"?
Does the game freeze / crash / black screen?
Or does the controller not send any input?
Make sure you don't press A on the gamepad after booting the game.

roxas10, 2022-01-16:
The controller won't send any input.

takoyaki99, 2022-01-16:
Did you update to the latest releases of Bloopair and Tiramisu?

roxas10, 2022-01-16:
I installed Tiramisu today, so should be updated

GaryOderNichts, 2022-01-16:
It works fine for me with an official switch pro.
Make sure to press A on the controller you actually want to use on the "Press A to start" screen.

roxas10, 2022-01-16:
Maybe was that I pressed A too much time? I tried again and now it works.

closed 2022-01-17 by GaryOderNichts