GaryOderNichts/Bloopair #53: Tiramisu Bloopair not working

Nottoknow2, 2022-01-18:
I am trying to pair a Xbox one s controller to the Wii U and it does not pair.

GaryOderNichts, 2022-01-19:
Is this a Series S/X controller or do you have the latest insider firmware on the controller? If so that's not compatible with Bloopair.
Otherwise make sure both the Wii U and Controller are in pairing mode and wait for a few seconds.

Nottoknow2, 2022-01-19:
> Is this a Series S/X controller or do you have the latest insider firmware on the controller? If so that's not compatible with Bloopair. Otherwise make sure both the Wii U and Controller are in pairing mode and wait for a few seconds.

No it is a Xbox one controller and i don't have the insider firmware

Nottoknow2, 2022-01-19:
Sorry to waste your time but it was a insider update i thought you had to enroll in it so i thought it was a normal update

closed 2022-01-19 by Nottoknow2

JohangHernandez, 2022-03-26:
For any others viewing this page trying to figure this out, you can roll back the firmware if you updated it.

[Check Out Here](https://answers.microsoft.com/en-us/xbox/forum/all/revertrollbackdowngrade-xbox-controller-firmware/25f07169-3244-4fe0-838c-949951b7c109?auth=1)