GaryOderNichts/Bloopair #56: Third party Wii U Pro controller and Wii motes won't synch

Konradovs, 2022-01-20:
Wii U Pro controller and Wii motes won't synch when the system boot with Tiramisu. I don't no exactly if it's a problem with bloopair or Tiramisu. If I boot in the normal system without SD it work without Problems. Ones they're synchronized they pair after booting even with Tiramisu without Problems.

GaryOderNichts, 2022-01-21:
Interesting. I can't reproduce this with any of my third-party controllers.
Can you try doing the following:
- Delete the `30_bloopair.rpx` file in the `wiiu/environments/tiramisu/modules/setup` folder
- Start your console with tiramisu
- Follow [this](https://en-americas-support.nintendo.com/app/answers/detail/a_id/1705/~/how-to-clear-all-syncs) to clear any paired controller synchronizations 
- Try pairing the controller again

Konradovs, 2022-01-22:
> Interesting. I can't reproduce this with any of my third-party controllers. Can you try doing the following:
>
> * Delete the `30_bloopair.rpx` file in the `wiiu/environments/tiramisu/modules/setup` [...]

Yep, that worked.

GaryOderNichts, 2022-01-22:
Okay now download the latest release from [here](https://github.com/GaryOderNichts/Bloopair/releases) and extract it to your SD Card.
Then try doing the last 3 steps from above again.

Konradovs, 2022-01-22:
Yep, doesn't work. Wii mote work but the pro controller from Sonviee doesn't

Konradovs, 2022-01-22:
Is it possible for someone like me to read the information from the controller and write it to an .in??

Konradovs, 2022-01-22:
But why they don't work? Do bloopair overwrite the system Bluetooth stuff? And even if you implement it in to bloopair does it work with nintendont? The original Nintendo Wii u pro controller works even with 4 players

GaryOderNichts, 2022-01-22:
Logs would help to figure out what the issue is.
Download the following file: [30_bloopair.zip](https://github.com/GaryOderNichts/Bloopair/files/7919150/30_bloopair.zip) and copy the `30_bloopair.rpx` inside of the `.zip` to `wiiu/environments/tiramisu/modules/setup` on your SD Card.
Then download the [UdpDebugReader.exe](https://github.com/dimok789/loadiine_gx2/raw/master/udp_debug_reader/UdpDebugReader.exe) and run it on a PC in the same network as your Wii U.
After that start your console with Tiramisu, and try to pair the controller. Then upload the logs the UdpDebugReader received here.

Konradovs, 2022-01-23:
Started logging to GeckoLog.txt.

[               main.cpp]                          main@L0056: Hello!
[               main.cpp]                          main@L0087: We need the kernel exploit to load the payload
[          gx2sploit.cpp]               DoKernelExploit@L0205: Running GX2Sploit
[          gx2sploit.cpp]               DoKernelExploit@L0272: GX2Sploit done
[             ElfUtils.c]                 LoadFileToMem@L0027: Loading file /vol/external01/wiiu/payload.elf.
[             ElfUtils.c]                   InstallMain@L0104: Copy section to 00FD0000 from 1062BDD8 (size: 111860)
[               main.cpp]                          main@L0095: New entrypoint at 00FD2294
[               main.cpp]                 start_wrapper@L0058: Hello from payload.elf multiloader
[               main.cpp]              get_all_payloads@L0139: Reading path wiiu/payloads/default
[               main.cpp]              get_all_payloads@L0150: default is valid!
[               main.cpp]              get_all_payloads@L0139: Reading path wiiu/payloads/nanddumper
[               main.cpp]              get_all_payloads@L0150: nanddumper is valid!
[               main.cpp]              get_all_payloads@L0139: Reading path wiiu/payloads/fw_img_loader
[               main.cpp]              get_all_payloads@L0150: fw_img_loader is valid!
[             ElfUtils.c]                 LoadFileToMem@L0028: Loading file /vol/external01/wiiu/payloads/default/payload.elf.
[             ElfUtils.c]                   InstallMain@L0145: Copy section to 00900000 from 10832A98 (size: 565964)
[               main.cpp]                 start_wrapper@L0079: loaded payload entrypoint at 00901B1C
[  ModuleDataFactory.cpp]                          load@L0105: Copy section .text 108BAFC0 -> 00E86700 [00F9D8CC] (1143244 bytes)
[  ModuleDataFactory.cpp]                          load@L0105: Copy section .rodata 109D21C0 -> 00F9D8CC [00FCC5A0] (191700 bytes)
[  ModuleDataFactory.cpp]                          load@L0105: Copy section .eh_frame 10A00EC0 -> 00FCC5A0 [00FF8360] (179648 bytes)
[  ModuleDataFactory.cpp]                          load@L0105: Copy section .data 10834300 -> 00FF8364 [00FFBFC0] (15452 bytes)
[  ModuleDataFactory.cpp]                          load@L0102: memset section .bss 00FFBFC4 [00FFEFD8] to 0 (12308 bytes)
[  ModuleDataFactory.cpp]                          load@L0112: Saved .bss section info. Location: 00FFBFC4 size: 00003014
[  ModuleDataFactory.cpp]                          load@L0127: Linking (1)... .text
[  ModuleDataFactory.cpp]                   linkSection@L0213: Found relocation section .rela.text
[  ModuleDataFactory.cpp]                   linkSection@L0257: done
[  ModuleDataFactory.cpp]                          load@L0127: Linking (3)... .rodata
[  ModuleDataFactory.cpp]                   linkSection@L0213: Found relocation section .rela.rodata
[  ModuleDataFactory.cpp]                   linkSection@L0257: done
[  ModuleDataFactory.cpp]                          load@L0127: Linking (5)... .eh_frame
[  ModuleDataFactory.cpp]                   linkSection@L0213: Found relocation section .rela.eh_frame
[  ModuleDataFactory.cpp]                   linkSection@L0257: done
[  ModuleDataFactory.cpp]                          load@L0127: Linking (7)... .data
[  ModuleDataFactory.cpp]                   linkSection@L0213: Found relocation section .rela.data
[  ModuleDataFactory.cpp]                   linkSection@L0257: done
[  ModuleDataFactory.cpp]                          load@L0127: Linking (9)... .bss
[  ModuleDataFactory.cpp]       getImportRelocationData@L0169: Found relocation section .rela.text
[  ModuleDataFactory.cpp]       getImportRelocationData@L0169: Found relocation section .rela.rodata
[  ModuleDataFactory.cpp]       getImportRelocationData@L0169: Found relocation section .rela.eh_frame
[  ModuleDataFactory.cpp]       getImportRelocationData@L0169: Found relocation section .rela.data
[  ModuleDataFactory.cpp]                          load@L0147: Saved entrypoint as 00E86718
[               main.cpp]                      do_start@L0136: Loaded module data
[               main.cpp]                      do_start@L0142: memset .bss 00FFBFC4 (12308)
[               main.cpp]                      do_start@L0151: Calling entrypoint at: 00E86718
[               main.cpp]                        _start@L0098: Freed 7 leaked memory blocks
Hello from mocha
Bye from mocha
Bloopair logging init
bta_search_callback called 1 0x121562e0
bta_search_callback called 4 (nil)
bta_search_callback called 1 0x121562e0
bta_search_callback called 4 (nil)
bta_search_callback called 1 0x121562e0
bta_search_callback called 4 (nil)
bta_search_callback called 1 0x121562e0
bta_search_callback called 4 (nil)
bta_search_callback called 1 0x121562e0
bta_search_callback called 4 (nil)
bta_search_callback called 1 0x121562e0
bta_search_callback called 4 (nil)

Konradovs, 2022-01-23:
sagt nicht viel. einfach nichts

Konradovs, 2022-01-23:
the first one again is when i try to synch the sonviee pro controller. just nothing.

te second one is the original wii u pro controller and the last one is a idont"know"name wii mote. they both work fine.

Bloopair logging init
bta_search_callback called 1 0x121562e0
bta_search_callback called 4 (nil)
bta_search_callback called 1 0x121562e0
bta_search_callback called 4 (nil)
bta_search_callback called 1 0x121562e0
bta_search_callback called 4 (nil)
bta_search_callback called 1 0x121562e0
bta_search_callback called 4 (nil)
bta_search_callback called 6 (nil)
bta_search_callback called 1 0x121562e0
bta_search_callback called 4 (nil)
bta_search_callback called 1 0x121562e0
bta_search_callback called 4 (nil)
bta_search_callback called 1 0x121562e0
bta_search_callback called 4 (nil)
bta_search_callback called 1 0x121562e0
bta_search_callback called 4 (nil)
bta_search_callback called 1 0x121562e0
bta_search_callback called 4 (nil)

bta_search_callback called 0 0x12156234
bta_search_callback called 1 0x121562e0
bta_search_callback called 2 0x12105c00
bta_search_callback called 4 (nil)
bta_sec_callback called 5 0x121562dc
bta_hh_di_sdp_callback called res: 0
report thread: received message 0
got vid 57E and pid 330
got name Nintendo RVL-CNT-01-UC status 0
btm_sec_execute_procedure_hook
bta_sec_callback called 2 0x12156258
bta_sec_callback called 3 0x1215625c
btm_sec_execute_procedure_hook
btm_sec_execute_procedure_hook
bta_hh_open_act handle 4 3
bta_hh_event called 2 0x1215639c
open event for handle 4
initController handle 4 magic b0 vid 57e pid 330
bta_hh_event called 11 0x121563c3
bta_hh_event called 16 0x121563d3
output request for handle 4 size 22, cmd: 0x16
writeDevInfo_hook (nil)
writeDevInfo_hook 0x11f3cd04
output request for handle 4 size 7, cmd: 0x17
output request for handle 4 size 2, cmd: 0x1A
output request for handle 4 size 2, cmd: 0x11
output request for handle 4 size 2, cmd: 0x15
output request for handle 4 size 3, cmd: 0x12
output request for handle 4 size 22, cmd: 0x16
output request for handle 4 size 22, cmd: 0x16
output request for handle 4 size 7, cmd: 0x17
output request for handle 4 size 22, cmd: 0x16
output request for handle 4 size 22, cmd: 0x16
output request for handle 4 size 22, cmd: 0x16
output request for handle 4 size 22, cmd: 0x16
output request for handle 4 size 7, cmd: 0x17
output request for handle 4 size 2, cmd: 0x13
output request for handle 4 size 2, cmd: 0x1A
output request for handle 4 size 22, cmd: 0x16
output request for handle 4 size 22, cmd: 0x16
output request for handle 4 size 22, cmd: 0x16
output request for handle 4 size 22, cmd: 0x16
output request for handle 4 size 22, cmd: 0x16
output request for handle 4 size 2, cmd: 0x15
output request for handle 4 size 2, cmd: 0x14
[... 152 lines cut ...]
output request for handle 3 size 22, cmd: 0x18
output request for handle 3 size 22, cmd: 0x18
output request for handle 3 size 22, cmd: 0x18
output request for handle 3 size 22, cmd: 0x18
output request for handle 3 size 22, cmd: 0x18
output request for handle 3 size 22, cmd: 0x18
output request for handle 3 size 22, cmd: 0x18
output request for handle 3 size 22, cmd: 0x18
output request for handle 3 size 22, cmd: 0x18
output request for handle 3 size 22, cmd: 0x18
output request for handle 3 size 22, cmd: 0x18
output request for handle 3 size 22, cmd: 0x18
output request for handle 3 size 22, cmd: 0x18
output request for handle 3 size 22, cmd: 0x18
output request for handle 3 size 22, cmd: 0x18
output request for handle 3 size 22, cmd: 0x18
output request for handle 3 size 2, cmd: 0x15
output request for handle 3 size 3, cmd: 0x12
bta_search_callback called 6 (nil)
output request for handle 3 size 2, cmd: 0x19
output request for handle 3 size 22, cmd: 0x16
output request for handle 3 size 22, cmd: 0x16
output request for handle 3 size 2, cmd: 0x14
output request for handle 3 size 2, cmd: 0x15
output request for handle 3 size 3, cmd: 0x12

GaryOderNichts, 2022-01-23:
I think I see what the issue is.
Can you upload logs from the following build: [30_bloopair.zip](https://github.com/GaryOderNichts/Bloopair/files/7920151/30_bloopair.zip)?

Konradovs, 2022-01-23:
jesus christ. what a log file

[               main.cpp]                          main@L0056: Hello!
[               main.cpp]                          main@L0087: We need the kernel exploit to load the payload
[          gx2sploit.cpp]               DoKernelExploit@L0205: Running GX2Sploit
[          gx2sploit.cpp]               DoKernelExploit@L0272: GX2Sploit done
[             ElfUtils.c]                 LoadFileToMem@L0027: Loading file /vol/external01/wiiu/payload.elf.
[             ElfUtils.c]                   InstallMain@L0104: Copy section to 00FD0000 from 1062BDD8 (size: 111860)
[               main.cpp]                          main@L0095: New entrypoint at 00FD2294
[               main.cpp]                 start_wrapper@L0058: Hello from payload.elf multiloader
[               main.cpp]              get_all_payloads@L0139: Reading path wiiu/payloads/default
[               main.cpp]              get_all_payloads@L0150: default is valid!
[               main.cpp]              get_all_payloads@L0139: Reading path wiiu/payloads/nanddumper
[               main.cpp]              get_all_payloads@L0150: nanddumper is valid!
[               main.cpp]              get_all_payloads@L0139: Reading path wiiu/payloads/fw_img_loader
[               main.cpp]              get_all_payloads@L0150: fw_img_loader is valid!
[             ElfUtils.c]                 LoadFileToMem@L0028: Loading file /vol/external01/wiiu/payloads/default/payload.elf.
[             ElfUtils.c]                   InstallMain@L0145: Copy section to 00900000 from 10832A98 (size: 565964)
[               main.cpp]                 start_wrapper@L0079: loaded payload entrypoint at 00901B1C
[  ModuleDataFactory.cpp]                          load@L0105: Copy section .text 108BAFC0 -> 00E86700 [00F9D8CC] (1143244 bytes)
[  ModuleDataFactory.cpp]                          load@L0105: Copy section .rodata 109D21C0 -> 00F9D8CC [00FCC5A0] (191700 bytes)
[  ModuleDataFactory.cpp]                          load@L0105: Copy section .eh_frame 10A00EC0 -> 00FCC5A0 [00FF8360] (179648 bytes)
[  ModuleDataFactory.cpp]                          load@L0105: Copy section .data 10834300 -> 00FF8364 [00FFBFC0] (15452 bytes)
[  ModuleDataFactory.cpp]                          load@L0102: memset section .bss 00FFBFC4 [00FFEFD8] to 0 (12308 bytes)
[  ModuleDataFactory.cpp]                          load@L0112: Saved .bss section info. Location: 00FFBFC4 size: 00003014
[  ModuleDataFactory.cpp]                          load@L0127: Linking (1)... .text
[  ModuleDataFactory.cpp]                   linkSection@L0213: Found relocation section .rela.text
[  ModuleDataFactory.cpp]                   linkSection@L0257: done
[  ModuleDataFactory.cpp]                          load@L0127: Linking (3)... .rodata
[  ModuleDataFactory.cpp]                   linkSection@L0213: Found relocation section .rela.rodata
[  ModuleDataFactory.cpp]                   linkSection@L0257: done
[  ModuleDataFactory.cpp]                          load@L0127: Linking (5)... .eh_frame
[  ModuleDataFactory.cpp]                   linkSection@L0213: Found relocation section .rela.eh_frame
[  ModuleDataFactory.cpp]                   linkSection@L0257: done
[  ModuleDataFactory.cpp]                          load@L0127: Linking (7)... .data
[  ModuleDataFactory.cpp]                   linkSection@L0213: Found relocation section .rela.data
[  ModuleDataFactory.cpp]                   linkSection@L0257: done
[  ModuleDataFactory.cpp]                          load@L0127: Linking (9)... .bss
[  ModuleDataFactory.cpp]       getImportRelocationData@L0169: Found relocation section .rela.text
[  ModuleDataFactory.cpp]       getImportRelocationData@L0169: Found relocation section .rela.rodata
[  ModuleDataFactory.cpp]       getImportRelocationData@L0169: Found relocation section .rela.eh_frame
[  ModuleDataFactory.cpp]       getImportRelocationData@L0169: Found relocation section .rela.data
[  ModuleDataFactory.cpp]                          load@L0147: Saved entrypoint as 00E86718
[               main.cpp]                      do_start@L0136: Loaded module data
[               main.cpp]                      do_start@L0142: memset .bss 00FFBFC4 (12308)
[               main.cpp]                      do_start@L0151: Calling entrypoint at: 00E86718
[               main.cpp]                        _start@L0098: Freed 7 leaked memory blocks
Hello from mocha
Bye from mocha
Bloopair logging init
bta_search_callback called 0 0x12156234
BTA_DM_INQ_RES_EVT: dev_class 280424
bta_search_callback called 0 0x12156234
BTA_DM_INQ_RES_EVT: dev_class 08043c
bta_search_callback called 0 0x12156234
BTA_DM_INQ_RES_EVT: dev_class 280424
bta_search_callback called 0 0x12156234
BTA_DM_INQ_RES_EVT: dev_class 280424
bta_search_callback called 0 0x12156234
BTA_DM_INQ_RES_EVT: dev_class 08043c
bta_search_callback called 0 0x12156234
BTA_DM_INQ_RES_EVT: dev_class 08043c
bta_search_callback called 0 0x12156234
BTA_DM_INQ_RES_EVT: dev_class 280424
bta_search_callback called 0 0x12156234
BTA_DM_INQ_RES_EVT: dev_class 08043c
bta_search_callback called 0 0x12156234
BTA_DM_INQ_RES_EVT: dev_class 280424
bta_search_callback called 0 0x12156234
BTA_DM_INQ_RES_EVT: dev_class 08043c
[... 1025 lines cut ...]
bta_search_callback called 2 0x12104580
[TV]Samsung LED55 is non official, replacing name...
bta_search_callback called 4 (nil)
bta_sec_callback called 5 0x121562dc
bta_hh_di_sdp_callback called res: 0
report thread: received message 0
[... cut ...]
bta_search_callback called 0 0x12156234
BTA_DM_INQ_RES_EVT: dev_class 280424
bta_search_callback called 0 0x12156234
BTA_DM_INQ_RES_EVT: dev_class 08043c
bta_search_callback called 0 0x12156234
BTA_DM_INQ_RES_EVT: dev_class 280424
bta_search_callback called 0 0x12156234
BTA_DM_INQ_RES_EVT: dev_class 08043c
bta_search_callback called 0 0x12156234
BTA_DM_INQ_RES_EVT: dev_class 08043c
bta_search_callback called 0 0x12156234
BTA_DM_INQ_RES_EVT: dev_class 280424
bta_search_callback called 0 0x12156234
BTA_DM_INQ_RES_EVT: dev_class 08043c

Konradovs, 2022-01-23:
i feel naked bro

Konradovs, 2022-01-23:
I guess something is clashing?

GaryOderNichts, 2022-01-23:
I put your logs in expandable blocks, to not fully clutter the thread.
The issue seems to be that your third-party controller doesn't have a valid device class, Bloopair is looking for.
The device class is used to check if that Bluetooth device is actually a controller and not some random Bluetooth headset.
The Wii U itself doesn't care about the device class.
I'll need to find a proper solution to fix this.

Your log above picked about a lot of other devices. Can you try getting logs from this build? This should also allow pairing the controller.
[30_bloopair.zip](https://github.com/GaryOderNichts/Bloopair/files/7920262/30_bloopair.zip)

Konradovs, 2022-01-23:
thank you. finaly i found some space. here we go

Bloopair logging init
bta_search_callback called 0 0x12156234
BTA_DM_INQ_RES_EVT: dev_class 002504
bta_search_callback called 1 0x121562e0
bta_search_callback called 2 0x12100e80
bta_search_callback called 4 (nil)
bta_sec_callback called 5 0x121562dc
bta_hh_di_sdp_callback called res: 0
report thread: received message 0
got vid FFFF and pid FFFF
got name Nintendo RVL-CNT-01-UC status 0
btm_sec_execute_procedure_hook
bta_sec_callback called 2 0x12156258
bta_sec_callback called 3 0x1215625c
btm_sec_execute_procedure_hook
btm_sec_execute_procedure_hook
bta_hh_open_act handle 5 3
bta_hh_event called 2 0x1215639c
open event for handle 5
initController handle 5 magic b0 vid ffff pid ffff
bta_hh_event called 11 0x121563c3
output request for handle 5 size 22, cmd: 0x16
bta_hh_event called 16 0x121563d3
writeDevInfo_hook (nil)
output request for handle 5 size 7, cmd: 0x17
output request for handle 5 size 2, cmd: 0x1A
output request for handle 5 size 2, cmd: 0x11
output request for handle 5 size 2, cmd: 0x15
output request for handle 5 size 3, cmd: 0x12
output request for handle 5 size 22, cmd: 0x16
output request for handle 5 size 22, cmd: 0x16
output request for handle 5 size 7, cmd: 0x17
output request for handle 5 size 22, cmd: 0x16
output request for handle 5 size 22, cmd: 0x16
writeDevInfo_hook 0x11f3cd04
output request for handle 5 size 22, cmd: 0x16
output request for handle 5 size 22, cmd: 0x16
output request for handle 5 size 7, cmd: 0x17
output request for handle 5 size 2, cmd: 0x13
output request for handle 5 size 2, cmd: 0x1A
output request for handle 5 size 22, cmd: 0x16
output request for handle 5 size 22, cmd: 0x16
output request for handle 5 size 22, cmd: 0x16
output request for handle 5 size 22, cmd: 0x16
output request for handle 5 size 22, cmd: 0x16
output request for handle 5 size 2, cmd: 0x15
output request for handle 5 size 2, cmd: 0x14
output request for handle 5 size 22, cmd: 0x16
output request for handle 5 size 22, cmd: 0x16
output request for handle 5 size 22, cmd: 0x16
output request for handle 5 size 2, cmd: 0x19
output request for handle 5 size 2, cmd: 0x15
output request for handle 5 size 22, cmd: 0x16
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18
output request for handle 5 size 22, cmd: 0x18

GaryOderNichts, 2022-01-23:
Hmm the device class seems fine.
Does pairing the controller work with this build?

Konradovs, 2022-01-23:
Yes the pairing works fine thank you. Is it a older build?

GaryOderNichts, 2022-01-23:
It's just a test build, which currently breaks support for most Bloopair compatible controllers.
I'll see if I can fix this tomorrow, thanks for all the testing so far!

Konradovs, 2022-01-24:
My Wii u doesn't shot down since your test release. Gamepad backlights are still on but the screen is black. The Wii U powerled is still blue and the fan is still working and also the USB accessories but no picture. I have to press the powerbutton on the Wii u for longer then 4 seconds. Could it be ur test build?

GaryOderNichts, 2022-01-24:
Yeah, that's because of the debug builds. You don't have to worry about it.

GaryOderNichts, 2022-01-26:
Sorry, I'm not sure why device class filtering doesn't work properly with some limited devices.
I'm a bit busy at the moment and will see what I can do to fix this in the future.

Konradovs, 2022-01-26:
No probs. I'm good with your actual release and my workaround works for now. Thank you

asc82, 2023-02-03:
The exact same issue also applies to the Wii Balance Board, it pairs fine without the plugin installed.

GaryOderNichts, 2023-02-05:
This should hopefully be fixed with the following build: [Bloopair_nightly.zip](https://nightly.link/GaryOderNichts/Bloopair/workflows/ci/main/Bloopair_nightly.zip).
Let me know if you still experience any issues.

GaryOderNichts, 2023-02-06:
Should be fixed with version 0.6.2.

closed 2023-02-06 by GaryOderNichts