GaryOderNichts/Bloopair #58: Thumbstick controls sometimes not working on third party Switch Pro

Borko-Jovanovic, 2022-04-06:
Often when I turn on my paired third party Switch Pro controller thumbstick controls are not working. 

Currently I resolve it by turning my controller on and off few times while moving one of the thumbsticks.

albertowd, 2022-06-20:
Same problem here, I have 3 3rd party controllers, some times it gets the sticks right, some times I cannot use the sticks.
Only the sticks, all other buttons are working correctly.

Can I set my app to debug some lines and paste here to help?

Dremii, 2022-08-09:
Same here my dudes

GaryOderNichts, 2022-08-09:
Sounds like the controllers might send invalid stick data in the beginning which is used for calibrating.
Can you try doing the following after the controller is connected: https://en-americas-support.nintendo.com/app/answers/detail/a_id/1107/~/how-to-reset-the-l-and-r-control-sticks-to-a-neutral-position-on-the-wii-u-pro

Dremii, 2022-08-09:
Thanks so much

closed 2025-04-07 by GaryOderNichts