GaryOderNichts/Bloopair #59: Can't my Xbox/Ps4 Controller even with Tiramisu!

SamuraiAdam, 2022-04-20:
issue is in title.

GaryOderNichts, 2022-04-22:
A few more details would be good.
Is your Wii U on the controller synchronization screen?
Are both controllers in pairing mode?
What version of the Xbox Controller are you using?
Can you test if Bloopair is running by starting the DS3 pairing menu from the HBL? Should give you an error if Bloopair isn't running.

GaryOderNichts, 2022-04-26:
If your Xbox Controller is on the latest firmware version, it's not supported by Bloopair.
Take a look [here](https://answers.microsoft.com/en-us/xbox/forum/all/revertrollbackdowngrade-xbox-controller-firmware/25f07169-3244-4fe0-838c-949951b7c109) or [here](https://github.com/ndeadly/MissionControl/issues/370#issuecomment-1089628024) to downgrade the firmware.

closed 2022-10-02 by GaryOderNichts