GaryOderNichts/Bloopair #62: Audio out PS4 controller?

mwhart88, 2022-04-28:
Would it be possible to somehow get headphone support working alongside the PS4 controller? I have my PS4 dualshock controller connected and the controller itself has the 3.5 mm Jack but no sound comes out through headphones and is still inputted thru the TV. Is there any way to patch or mod this to allow to  sound pass thru headphones?

gitdrug, 2022-10-23:
This would also be good feature on other controllers like XB1, and Dulesense.

rusa99ec, 2022-12-26:
I don't think so. Someone correct me if Im wrong but as far as I know Wii U bluetooth stack doesnt support audio, so no chance of that. Also remember that Bloopair emulates a Wii U Pro Controller and that controller doesnt have that feature.