GaryOderNichts/Bloopair #67: Possible Wired Controller support?

dude98, 2022-07-17:
i have a wired only xbox one controller, and multiple xbox 360 controllers, would it be possible to support wired controllers along with the xbox 360 wireless receiver? the 360 didn't use bluetooth so even the wireless 360 controller won't connect. i would use hid to vpad but xinput doesn't work directly with it, and the network client is very laggy when entering a game since i don't use ethernet on my wii u.

mariomadproductions, 2022-09-27:
HID to VPAD is inherently laggy, according to Maschell. Also it doesn't seem to be maintained anymore. So having wired support in bloopair would be useful (plus I think adding new support for new controllers that use USB is easier than with bluetooth).

Invictaz, 2022-10-02:
Second this

GaryOderNichts, 2022-10-02:
Integrating USB support into Bloopair just because HIDtoVPAD is unmaintained, doesn't seem like a fair point to me.
It would make more sense to update HIDtoVPAD with support for the new plugin system which is part of Aroma.

Invictaz, 2022-10-02:
> Integrating USB support into Bloopair just because HIDtoVPAD is unmaintained, doesn't seem like a fair point to me. It would make more sense to update HIDtoVPAD with support for the new plugin system which is part of Aroma.

Would it work on vWii then in Aroma? Because Fakemote from Xerpi only supports a limited amount of controllers?

GaryOderNichts, 2022-10-02:
> Would it work on vWii then in Aroma? Because Fakemote from Xerpi only supports a limited amount of controllers? @GaryOderNichts

Nothing which runs on the Wii U side will run while in vWii.

Soundtoxin, 2023-02-15:
USB controller support would be great for using an Open-Frame1 controller with Smash 4 over USB instead of needing a custom USB-C to GC cable + Wii U GC adapter (like you can on Switch already). 

If this is not fit for inclusion in Bloopair, maybe pinning and closing the issue would be a good idea for other people who come by curious about it. The open issue got me excited that maybe it was a possibility.

Yeetov, 2023-12-16:
> Integrating USB support into Bloopair just because HIDtoVPAD is unmaintained, doesn't seem like a fair point to me. It would make more sense to update HIDtoVPAD with support for the new plugin system which is part of Aroma.

This would be great for Aroma as there is no wired controller support with wired Xbox controllers, wired Nintendo switch controllers, GameCube adapter controllers and many more. This would be a game changer.

AlphaCraft9658, 2024-03-22:
Any updates on this?

Invictaz, 2024-03-22:
There is HID2vpad for Aroma. However not perfect we might as well have a USB input option somewhere. The Brook WiiU adapter emulates a WiiMote over usb with possibly a extension controller attached.

AlphaCraft9658, 2024-03-22:
> There is HID2vpad for Aroma. However not perfect we might as well have a USB input option somewhere. The Brook WiiU adapter emulates a WiiMote over usb with possibly a extension controller attached.

Can you send a link? The original HIDToVpad doesn't officially support Aroma.

Invictaz, 2024-03-22:
> > There is HID2vpad for Aroma. However not perfect we might as well have a USB input option somewhere. The Brook WiiU adapter emulates a WiiMote over usb with possibly a extension controller attached.
> 
> Can you send a link? The original HIDToVpad doesn't officially support Aroma afaik.

From the Aroma Discord: 

[hidtovpad.zip](https://github.com/GaryOderNichts/Bloopair/files/14728298/hidtovpad.zip)

closed 2026-09-30 by GaryOderNichts