GaryOderNichts/Bloopair #72: Xbox Controller supported firmware version

LeVraiRoiDHyrule, 2022-10-30:
Hi,

In the doc, it is said : "Microsoft Xbox One S/X Controller
Note: The latest firmware versions and all Series S/X Controllers are currently not supported due to missing Bluetooth LE support"

Could someone share more precision ? Up to what firmware version number Bloopair is compatible ? That will allow me to know if the fact that it doesn't work for me is due to the firmware version of my controller or something else.

Thanks in advance

GaryOderNichts, 2022-10-30:
I'm not sure exactly, since Microsoft doesn't provide proper release notes for those firmwares.
Bluetooth LE rollout started with insider builds last year and went into stable firmwares a few months later.
From looking at the version of my Xbox One controller I assume all versions starting with 5.XX.XXX.X will use BLE.

LeVraiRoiDHyrule, 2022-10-30:
Thanks for your answer.
Ok I see. All my controllers are on 5.XX.XXX.X. They probably using BLE. I downgraded to 3.1.1221.0 using this official method: https://support.xbox.com/en-US/help/hardware-network/accessories/controller-firmware-reversion, and it works now. I can connect correctly. But this is not optimal, it introduces some latency when playing on Xbox or PC using the dongle. But it allows to confirm that firmware was the issue. Thanks for the explaination. 

I guess the another is to wait if, maybe, in the future, you find a way to use it on Wii u as you said here : https://github.com/GaryOderNichts/Bloopair#to-do

Anyway thanks for your help and have a great day

closed 2022-11-01 by GaryOderNichts