GaryOderNichts/Bloopair #76: ZL/ZR don't work on the Nintendo Switch Pro Controller

CCF100, 2022-12-05:
I'm playing a NSMBU mod and the ZL and ZR buttons aren't responding :/

darkspace, 2023-01-06:
Same issue with GameSir-T4 Mini controller playing Super Mario 3D World, ZL doesn't work.  Haven't tested ZR but I assume it would also not function.  This controller is a third party Switch Pro controller.

GaryOderNichts, 2023-01-06:
There seem to be special shoulder button bits for joy-con. I wonder if some third-party Switch Pro Controllers set the wrong bits for the ZL/ZR buttons.
I'll look into it and try making a test build.

darkspace, 2023-01-07:
After further investigation it appears my controller is defective.  ZL and ZR do not work when paired to a Switch console either, and also do not work when USB connected to a Windows PC.  All other buttons and joysticks work, just not ZL and ZR.

closed 2024-06-04 by GaryOderNichts