GaryOderNichts/Bloopair #80: Can't connect a third party Switch Pro controller

Sistaaa, 2022-12-25:
Hi, I'm new to hacking a Wii U. I've installed Tiramisu and Bloopair, but I can't connect my third party Switch Pro controller (model MG26Z). Is there a way to fix this?

Thank you in advance.

GaryOderNichts, 2022-12-25:
Can you connect the controller to a PC or phone and check with which name it shows up?

Sistaaa, 2022-12-25:
There are a couple of ways to connect the controller.

1. With bluetooth, the name is "GamePadPlus V3"
2. With the receiver, the name is "Microsoft X-Box 360"
3. With the cable, the name is "Microsoft X-Box 360"

GaryOderNichts, 2022-12-25:
Please test if the following build works:
[30_bloopair.rpx.zip](https://github.com/GaryOderNichts/Bloopair/files/10300154/30_bloopair.rpx.zip)

Sistaaa, 2022-12-25:
Unfortunately, that didn't work. I guess that build was the same as I had.

GaryOderNichts, 2022-12-25:
That build had the `GamePadPlus V3` name added, so that Bloopair would identify the controller as a Switch controller.
Does the controller work on a Switch? Is there some special toggle which puts the controller into "Switch mode"?

Sistaaa, 2022-12-25:
Ah I see. I don't have a Switch, I use the controller with emulators, but it should work with Switch.

I found this online: https://ae01.alicdn.com/kf/H4b5cc5fea7414a068262f5ebe7f386599.jpg
With Android I use A+Home to enter in pairing mode, but for Switch it should be Y+Home. I just tried it with both builds, but without luck.

I searched further and found https://ae01.alicdn.com/kf/S62905ac28f984268bb70e6a2995beeaeq/Originele-Draadloze-Controller-Bluetooth-Voor-Nintendo-Schakelaar-Pro-Pc-PS3-Joystick-Gamepad-Nfc-Professionele-Vertragingsvrije-Gamepad.jpg
It looks like the model I have (MG26Z) only works with the receiver on Switch. So I plugged the receiver on my Wii U and it seems te connect (the lights om my controller stops blinking), but I can't use the controller.

Would it work if you add the "Microsoft X-Box 360" name to the build?

GaryOderNichts, 2022-12-25:
Ah, unfortunately Bloopair does not support wired controllers / receivers plugged in via USB.
If you can't connect the controller wirelessly to a Switch it won't work.

Sistaaa, 2022-12-25:
Oh no, that's such a pity. Just bought two of the controllers :(
Well, nothing to do about it, I can finally give up now, it gave me a headache.
Thank you so much for your time and help, appreciate it! 👍

closed 2024-06-04 by GaryOderNichts