GaryOderNichts/Bloopair #85: [Question] Stadia Controller compatibility

FranzXaver, 2023-01-19:
Now that we are able to enable blueetooh on the Google Stadia controller, are there any plans to add this controller?

FYI you can enable bluetooh on the Stadua controller here:
https://stadia.google.com/controller/index.html

GaryOderNichts, 2023-01-19:
Looks like the Stadia controller is using BLE which is not supported by Bloopair and most likely won't be supported in the future.

FranzXaver, 2023-01-19:
Thanks Gary, I am trying to find out if the bluetooth controller could maybe support BLE, but I can't find any specs. 
It should be a Broadcomm BCM20702 (2878D-WIN-B2), right?

Gius29, 2023-01-29:
I would be amazing to use Stadia controller with the wiiU

GaryOderNichts, 2026-10-04:
There are experimental Bloopair test builds available at https://github.com/GaryOderNichts/Bloopair/discussions/217 with support for the Stadia controller.

GaryOderNichts, 2026-10-25:
[Version 1.1.0](https://github.com/GaryOderNichts/Bloopair/releases/tag/v1.1.0) now supports the Stadia controller.

closed 2026-10-25 by GaryOderNichts