GaryOderNichts/Bloopair #86: Failed to read local bda (Make sure bloopair is active!) error happens and i tried everything

7H3-5P4N15H-0N3, 2023-01-21:
Before you guys tell me to

1. Use tiramisu
or
2. Get the latest version,

I did all of this requirements and still, on boot, this appears on screen.
`=== Bloopair USB Controller pairing`
`Connect a DualShock 3 using a USB cable to pair it`
`Press HOME to exit`
`Failed to read local bda (Make sure bloopair is active!)`
@GaryOderNichts or anyone else, pls tell me how 2 solve

GaryOderNichts, 2023-01-21:
You most likely did not follow the instructions properly and didn't copy the setup module to the right location. You're now only launching the Dualshock3 Pairing Menu without having the Bloopair module running.
As stated in the README:
> Copy the `30_bloopair.rpx` from the `.zip` file to the `modules/setup/` folder of your target environment on the SD Card.
> This would be `wiiu/environments/tiramisu/modules/setup/` for Tiramisu.

7H3-5P4N15H-0N3, 2023-01-22:
Ok il see if it works

closed 2023-02-11 by 7H3-5P4N15H-0N3

Natethebest20, 2023-02-16:
I'm having the same issue I did everything in the README but it still says the same thing