GaryOderNichts/Bloopair #94: After Bloopair update version 0.6.0, third-party Switch controllers can connect but not work

mhh996378373, 2023-02-21:
I use google translation, hope you can understand.

My GameSir T4Pro controller worked fine before in Switch mode (in Bloopair0.5.3 and all previous versions), after updating 0.6.0, when WIIU is connected, it shows that the connection is successful but the window does not display the battery icon, and any button of GameSir T4Pro does not work .

mhh996378373, 2023-02-21:
Fortunately, GameSir T4Pro can be switched to Xbox one mode, and it works perfectly in any version

moninox, 2023-03-17:
My clone pro controller works fine with 0.5.3 version but with 0.6.2 it shows that the connection is successful but the window does not display the battery icon, and the controller does not work .

joaonaweb, 2023-04-18:
I have the same problem as @mhh996378373 ! :/

ArmandoDrake, 2023-04-20:
Tested GameSir T4 mini with Bloopair 0.5.3 version and works fine, except vibration, it's not working for now.