GaryOderNichts/Bloopair #100: A quick question.

Walegy, 2023-04-06:
I have a strange question... Does Bloopair work with the Wavebird Gamecube controllers? i know it has a wireless usb dongle, as I am very curious if those sorts of things will ever get support on Bloopair, yet- I think it may work because the dongle does emit a signal to the controller itself, so that begs my question. Does the Wavebird work with Bloopair?

ThefakeSuperidol, 2023-07-22:
I'm not an expert, but look up something like joycons on wii u and there should be a yt video that shows off multiple controllers being paired with a dongle. I hope this helps.

Invictaz, 2024-02-24:
Wavebird uses a custom wireless protocol, no Bluetooth.

The only option to use a Wavebird would be via HID2Vpad and the WUP-028 adapter.

closed 2024-06-04 by GaryOderNichts