GaryOderNichts/Bloopair #108: NSO N64 controller has greater input lag than the Gamepad

al797, 2023-08-19:
I own a NSO N64 controller but don't own a Switch so unfortunately I can't say this is an issue with the controller itself.
On the Wii U, at least, I can tell there's noticeably a higher input lag compared to the Wii U Gamepad. Is this something to be expected due to how Bloopair handles these officially unsupported controllers or is it something that might actually be improved?

SvenCreg, 2025-03-21:
Hello, I just wanted to add that the lag on the NSO Switch controller is pretty bad. Sometimes the buttons do not respond at all.  I'm not sure if there is any way to improve it.

GaryOderNichts, 2026-10-04:
@HVC-001 I created a test branch which lowers the report interval to 5 ms. Can you test if the following build improves the input lag? You can download a test build here: https://nightly.link/GaryOderNichts/Bloopair/workflows/ci/test%2Ffaster_report_thread/Bloopair_nightly.zip

Bloopair used to directly translate the incoming reports and immediately pass them through to the PPC side. However this caused issues with some controllers that only sent reports when buttons were pressed, causing the PPC side to think the controllers had been disconnected. There were also some controllers which were sending reports extremely fast, causing the report queue to overflow.

So now Bloopair is collecting reports from all controllers and sending them to the PPC side at once. However, I agree, this is not ideal for input lag. The additional button mapping translation probably also doesn't help.

HVC-001, 2026-10-04:
Unfortunately @GaryOderNichts, there doesn't seem to be any improvement with the new nightly build.

For reference, this is the methodology I've been using to measure the input lag:

- **Monitor**: Samsung Odyssey G3, 60 Hz, response time set to "Fastest".
- I record a video at **240 FPS**, so every 4 frames in the video correspond to approximately 1 frame on the actual display.
- The game I'm testing with is **Taiko no Tatsujin Wii U Version!**, using the base disc version with no updates. The game runs at **60 FPS**.
- I'm measuring the time between pressing the button and the taiko note being registered on screen.

It's obviously not the most precise or professional way of measuring input latency, since there are quite a few variables involved, but I think it's good enough to get a rough idea of the differences.

My results are:

**No Bloopair**

- **Wii U GamePad:** ~12 video frames (~3 game frames of input lag)
- **Wiimote:** ~12 video frames (~3 game frames of input lag)

**Bloopair 1.0.4**

- **DualShock 4:** ~18 video frames (~4–5 game frames of input lag)
- **Switch Pro Controller:** ~24 video frames (~6  game frames of input lag) -> _I think we're seeing the infamous inherent lag of Switch controllers in action here comparing with the DualShock._

**Bloopair Nightly (test/faster_report_thread)**

- **DualShock 4:** ~18 video frames (~4–5  game frames of input lag)
- **Switch Pro Controller:** ~24 video frames (~6  game frames of input lag)

So, at least with these controllers and this test, lowering the report interval to 5 ms doesn't seem to make a measurable difference.

Also, shortly after my previous message, I recompiled Bloopair myself with REPORT_INTERVAL set to 1 instead of 5. I did this somewhat recklessly, assuming it would probably break something, since there presumably has to be a reason why the interval was set to that particular value in the first place. Interestingly, I didn't see an obvious improvement in input lag there either.

So, basically, twice the input lag with Switch controllers. Of course, if we use a 30 FPS game like TPHD, or even more so an N64 Virtual Console game with its own atrocious input lag, which is the original complaint of this issue, it will _feel_ twice as bad.

> Bloopair used to directly translate the incoming reports and immediately pass them through to the PPC side.

Is the build of the version that worked this way from before Aroma? If not, I'd like to test it.

Thanks for looking into this!

GaryOderNichts, 2026-10-04:
> Is the build of the version that worked this way from before Aroma? If not, I'd like to test it.

Looking back, this was changed with commit https://github.com/GaryOderNichts/Bloopair/commit/22a7bad3218c267db8f3880ec9670b2ee1cb5c43 in [v0.2.0](https://github.com/GaryOderNichts/Bloopair/releases/tag/v0.2.0). So unfortunately this predates both Aroma and Tiramisu.

I can probably create a build which does something similar again though. A build without button remapping would also be interesting.

I definitely appreciate your testing though @HVC-001!

GaryOderNichts, 2026-10-04:
@HVC-001 Here are two more builds for testing:
- https://github.com/GaryOderNichts/Bloopair/actions/runs/37212417654/artifacts/11307113562
  This one disables the button and stick remapping code. I don't think this makes much of a difference, but would be interesting to see.
- https://github.com/GaryOderNichts/Bloopair/actions/runs/37213334466/artifacts/11306529387
  This one completely bypasses the reporting thread and sends inputs directly to the PPC side once they've been received and translated to the Pro Controller format.

Note that both builds only work with the Switch Pro Controller.
It would also be interesting to see how much delay a Wii Remote has, when using Bloopair. Just in case any of the Bloopair patches cause additional delay.

HVC-001, 2026-10-04:
Sadly, I'm still counting 6 frames in both builds when using the Switch Pro Controller. 😓 But I don't see any additional frames when using the Wii Remote, it's the same with or without Bloopair.

GaryOderNichts, 2026-10-04:
Unfortunately I don't think there's much Bloopair can do here then. The only other things Bloopair is doing for each report is calibrating the analog sticks, translating the button bits, and encrypting the report. However, if the other builds didn't make any difference, I don't think this is affecting anything.

Thank you for testing this though!

HVC-001, 2026-10-04:
Yes, I think that's how it is. At most, Bloopair adds that extra frame seen with the DualShock 4, which is a good trade-off for gaining compatibility.

It's just that the Switch Pro Controller is terrible, as has been well documented (https://gamepadla.com/nintendo-switch-pro-controller.html), which makes me incredibly sad because it's so much more comfortable in my hands than the Wii or Wii U Pro Controller. And it makes me even sadder when I think about the NES/SNES/N64 controllers, which would be absolutely perfect for the Virtual Console.

But hey, apparently the Switch 2 Pro Controller is better (https://gamepadla.com/nintendo-switch-2-pro-controller.html), although still not perfect. For now, I only have the NSO GameCube controller, which I assume is basically a Pro 2, so now that you've added BLE support, I should give it a try.

Thanks for everything!

ndeadly, 2026-10-06:
@GaryOderNichts are you aware of Switch controller firmware behaving differently depending on the name of the Bluetooth host adapter? People have reported issues with input delay and disconnections on platforms like Steam Deck which are resolved by setting the adapter name to match one of the expected prefixes. This thread goes into a bit of detail on it https://github.com/DanielOgorchock/linux/issues/33#issuecomment-2849798054.

GaryOderNichts, 2026-10-06:
> are you aware of Switch controller firmware behaving differently depending on the name of the Bluetooth host adapter?

I was not aware, what?! I just tested it by setting the name to "Nintendo Switch" and it did actually change the input report.

@HVC-001 Can you test if this build changes anything? https://github.com/GaryOderNichts/Bloopair/actions/runs/37482537961/artifacts/11421513239

I changed the device name to "Nintendo Wii U" as opposed to just "Wii" (Yes, the Wii U names itself "Wii"). I always assumed that was just a copy-paste mistake, now I wonder if there's similar shenanigans going on in the Wii Remote firmware.

nfp0, 2026-10-09:
@HVC-001 I am pretty sure the guys at Gamepadla did not test the Switch controllers in high input report rate mode.

HVC-001, 2026-10-09:
> Can you test if this build changes anything?

@GaryOderNichts I'm now counting 7 to 8 frames... 😳
Wii Remote still at 3 frames with that build.

HVC-001, 2026-10-09:
> 15\. I am pretty sure the guys at Gamepadla did not test the Switch controllers in high input report rate mode.

I don't really know, but something is definitely going on with the Switch or its controllers. During the Wii and Wii U era, there was very little input lag in the Taiko no Tatsujin games, just the 3 frames we're discussing in this thread.

When I bought a Nintendo Switch along with Taiko no Tatsujin: Nintendo Switch Version!, I noticed the input lag immediately. It's more than twice as bad on Switch, whether you're using Joy-Cons or a Pro Controller, making the game impossible for me to play. Initially, I assumed Bandai Namco had messed up something in their software, but then you start noticing input lag everywhere on the Nintendo Switch. Even the system menu itself has far more input lag than the Wii U. And, of course, it's night and day compared to the DS or 3DS.

The same goes for motion controls. Playing Splatoon on Wii U with the GamePad's gyro and then switching to Splatoon 2 on Switch, the difference in responsiveness is immediately apparent. I always assumed it was either the controller itself or developers buffering frames for performance reasons. The same goes for Smash Ultimate, [which has the highest input lag of any game in the series to date](https://www.polygon.com/2018/12/14/18140814/super-smash-bros-ultimate-input-lag-latency-feel/).

There are other games, such as the M2 ShotTriggers series, that consistently have 1–2 more frames of input lag than their counterparts on PS4.

So, I don't know. The input lag on the Nintendo Switch and its controllers has me completely baffled.

GaryOderNichts, 2026-10-09:
> [@GaryOderNichts](https://github.com/GaryOderNichts) I'm now counting 7 to 8 frames... 😳
> Wii Remote still at 3 frames with that build.

Hmm so it's worse than before? I wonder if the Wii U can't keep up with all of the reports the Switch Pro Controller sends.

nfp0, 2026-10-09:
@HVC-001 All of that might be true, but it's all subjective until an apples to apples comparison can be done with actual measurements.
And one thing is known: The Switch Pro controller has 2 report modes. A slow and a fast one. And the fast one needs to be activated by a sequence of instructions as described [here](https://github.com/dekuNukem/Nintendo_Switch_Reverse_Engineering/blob/master/bluetooth_hid_notes.md#input-0x30).
I still think Gamepadla did test with only a simple Bluetooth connection on Windows, which does not activate the fast mode.

The Linux driver activates the fast mode, and that would be a better test of the controller's full capability.

GaryOderNichts, 2026-10-12:
@HVC-001 I have two more builds to test:
- https://github.com/GaryOderNichts/Bloopair/actions/runs/38309229403/artifacts/11774057148
  The Wii U usually changes the controllers to sniff mode with an 8ms interval, and a 16ms interval if more than 4 controllers are connected. This build keeps the controller always in active mode.
- https://github.com/GaryOderNichts/Bloopair/actions/runs/38309288632/artifacts/11773812608
  This does the same as above, while additionally also changing the device name to "Nintendo Wii U" for the Switch Pro controller.

HVC-001, 2026-10-13:
@GaryOderNichts I think there's an improvement in the first build, I'm counting 5 frames! However, with the second build, I'm counting 6 frames again. When we're talking about differences of just 1 frame, it can sometimes be tricky to tell with a method as rudimentary as mine. The difference becomes more evident when comparing the Wiimote's 3 frames to the Switch Pro's 6. I've measured the first build several times, though, and I consistently count 5 frames. Still, I wish someone with better equipment could take precise measurements to help you compare each build way more accurately.

nfp0, 2026-10-13:
@HVC-001 Sorry if this is a bit off-topic, but since I am also doing input lag measurements with a high speed video (480 fps), I am wondering what is your procedure after you capture the video.
Did you automate it somehow?

I personally have been opening MPV and use the frame advance key (.) to count them manually, but that gets very repetitive when repeating over many samples, so I was wondering if there is a better way.

Also, how many samples do you count for the average?

HVC-001, 2026-10-14:
@nfp0 Exactly the same method, using MPV and going frame by frame. At 240 Hz, it’s already tedious, so I imagine it’s even more so at 480 Hz 😅 ([some people are doing it at 1000 Hz!](https://www.youtube.com/watch?v=ZU7q7q6H7Y4)) I have no way to automate it. I test 10 samples, sometimes more just to make sure. I use Taiko no Tatsujin because it’s visually very easy to tell when a frame has changed by watching the movement of the note chart.

nfp0, 2026-10-14:
@HVC-001 Aw, I was hopeful you had some automation tips for me, but I guess manual counting it is! 😛 
Yeah, I also count 10 samples usually.