GaryOderNichts/Bloopair #111: Super Famicom Online Controller ZR not working

IceBlueLugia, 2023-09-08:
Pretty much summed up in the title. I have the Super Famicom Switch Online controller connected via Bloopair and the ZR button just doesn’t do anything; it’s brand new so I doubt there’s any issue. ZL works fine oddly enough. Is this intentional? ZR feels more instinctual to press for the Virtual Console menu

GaryOderNichts, 2023-09-08:
There's currently no special handling for these types of controllers so it should just be handled by the basic reports.
Even then ZR should work if all other buttons are working.

IceBlueLugia, 2023-09-11:
Yeah I’m really not sure. Controller isn’t defective or anything, when I connect it to the switch ZR works just fine. Not a huge deal I guess, but a little unfortunate. Wondering if the SNES Online one would have the issue too; maybe I’ll get one eventually for two player games

IceBlueLugia, 2023-09-15:
Hi, I tried a friend’s SNES Online controller and the same thing happens. Everything works except ZR. And yeah, the ZR button on that controller works fine on my Switch. Guessing there’s some weird bug here that’s exclusive to this controller I totally get it if this is sort of a low priority issue for you since I’d assume you have to write code for another new controller to address this, but I’d appreciate if you could look into this whenever you get the chance. Thanks for the awesome software man, this has really helped turned my Wii U into my retro machine. Have a good one :)

daveq86, 2023-10-14:
I can second this. The ZR on my PAL NSO SNES controller does not work. (This button is needed to bring up the emulation-menu during any virtual game.