GaryOderNichts/Bloopair #113: LEFT STICK NO WORKING UPPER AND RIGHT ON FAKE PRO CONTROLLER SWITCH

ramsaynieve92, 2023-09-20:
Hello

LEFT STICK NO WORKING UPPER AND RIGHT ON FAKE PRO CONTROLLER SWITCH.

sync
all work except left stick up and right

game cube pro controller fake for switch and pc

ramsaynieve92, 2023-09-22:
@GaryOderNichts  any help my dude? thanks for ur work

Sianure06, 2024-01-14:
I have the same problem with my switch plate controller from mobile gaming corps. Both sticks doesn't works at all, the left one only moves to the left and the right one is useless.

GaryOderNichts, 2024-01-14:
I think this might be caused by Bloopair trying to read calibration data from the controller, but the fake controllers aren't returning valid data.
Can you try [Version 0.5.3](https://github.com/GaryOderNichts/Bloopair/releases/tag/v0.5.3), which doesn't implement calibration yet?

Izooc, 2024-03-03:
> I think this might be caused by Bloopair trying to read calibration data from the controller, but the fake controllers aren't returning valid data. Can you try [Version 0.5.3](https://github.com/GaryOderNichts/Bloopair/releases/tag/v0.5.3), which doesn't implement calibration yet?

I tried 0.5.3 and now the joystick just doesnt work at all.

UnintelligentSoul, 2025-02-16:
I had this same issue with a 8BitDo Pro 2 Bluetooth gamepad when using it with the default Bluetooth mode.  I switched it to xinput mode and it now works fine.

GaryOderNichts, 2025-04-07:
There were some calibration issues in previous versions which should be fixed with release v1.0.3.