GaryOderNichts/Bloopair #115: Random drift  on official switch pro controller

daveq86, 2023-10-14:
Random “stickdrift” on Switch Pro controllers during gameplay. (Never experienced drift with these controllers on NSwitch, tried with 2 different controllers) And randomly turned off, leading the character on screen to move left/right indefinitely.

varies from once every hour, to 3 times in 5 minutes.

Work-a-round: turn off controller (with short press on the sync button) followed by any random button press, to resync the controller.

F.y.i: Switch Pro Controllers are updated to latest version

(other than that: GREAT product! Love it! Keep up the good work!)

drittich, 2025-02-04:
Same here, it either works great, or goes completely wonky. Usually happens once or twice per session.

GaryOderNichts, 2025-04-07:
There were some calibration issues in previous versions which should be fixed with release v1.0.3.

closed 2025-04-07 by GaryOderNichts