GaryOderNichts/Bloopair #127: [Feature request] Support for generic USB controllers

AlphaCraft9658, 2024-03-22:
I am not sure whether this could be possible, but could support for generic XInput or DInput USB-Controllers be added? The protocol exists and has implementations in open-source projects like the Linux Kernel (or other Linux packages, idk). So could it be implemented, so that generic XInput/DInput controller signals can be translated to Wii U Pro Controller data the console can understand. But as I said, I don't know whether it's possible to implement or whether the effort would be too high.

Invictaz, 2024-03-22:
We have stuff like https://github.com/isage/vixen which might be able to be ported to the ARM side of the WiiU. But you would have to ask @GaryOderNichts

AlphaCraft9658, 2024-03-23:
> We have stuff like https://github.com/isage/vixen which might be able to be ported to the ARM side of the WiiU. But you would have to ask @GaryOderNichts 

The ARM side of the Wii U? Doesn't it use the PowerPC architecture?

GaryOderNichts, 2024-03-23:
Bloopair already supports the Xbox One S/X Controller. Support for wired controllers is currently not planned.

closed 2024-03-23 by GaryOderNichts

AlphaCraft9658, 2024-03-23:
> Bloopair already supports the Xbox One S/X Controller. Support for wired controllers is currently not planned.

We have two controllers which are connected with a WiFi dongle. It would be pretty useful if there was something to provide compatibility.

Invictaz, 2024-03-23:
The WiFi dongle that the Xbox Series Controllers use might be able to get supported but nobody made it work. There is not even a single hid dump of it.