GaryOderNichts/Bloopair #138: Single Joycon and 3rd Party Pro controller not syncing

datboi2008, 2024-06-23:
I have read up on other issues, and I have seen several others with third party switch pro controllers not pairing. This one is a PDP branded pro controller, but I know that with the current version 3rd party pro controllers don't work. The real issue is the Joycon. Im using a first Party Joycon, but It will not pair. My PS4 controller pairs fine and I know xbox controllers don't work. I cannot find anyone else with this issue.

GaryOderNichts, 2024-08-16:
If it's a newer third-party Switch Pro Controller, this should be fixed in [version 1.0.1](https://github.com/GaryOderNichts/Bloopair/releases/tag/v1.0.1). As for the Joycon try connecting it as the first controller and make sure it's fully charged.

theotheroracle, 2025-02-21:
i have a left joycons that pairs fine, and a right joycon that doesnt pair, as far as I know its an official one, but they're both second hand . its fully charged, and I'm trying to pair it with no other controllers connected, I'm running bloopair 1.0.2, any suggestions ?