GaryOderNichts/Bloopair #143: [Regression] New Bloopair version breaks Balance Board support

VGmaster670, 2024-07-24:
What it says on the title, trying to pair a balance board to a console with Bloopair 1.0 installed causes the console to reject the connection, but after uninstalling Bloopair and rebooting the console, it works fine. (The balance board was worked fine on previous versions)

GaryOderNichts, 2024-08-12:
Hmm that is weird. Nothing should've changed for official controllers with Bloopair 1.0.0. Can you try using v0.6.2?

VGmaster670, 2024-08-13:
I'm travelling right now, but I get back in a bit under a week, so I'll do
so then 👍

VGmaster670, 2024-08-22:
It... seems to work fine now? Not sure why it rejected the connection before (although it still seems to reject the connection on the home menu, but that might be intended behavior), but it connects in Wii Fit U now with or without Bloopair installed. Guess I'll close the issue now

closed 2024-08-22 by VGmaster670