GaryOderNichts/Bloopair #160: Wii U finds generic PS3 controller, but doesn't work.

FERFAIL-TXZ, 2024-12-08:
I am very sure my generic PS3 controller works, I tested it in my PC, no issues.

The Wii U finds the PS3 controller without any issues, even when I disconnect the controller cable, the controller red light appears as the second player, I assume that it should be like that.
However, the controller doesn't work. Touching any buttons doesn't do anything.

I just recently hacked my Wii U using a guide, but I am not very experienced about that, I just followed instructions and that's it. So, please if its possible to give a solution, give it more detailed.
[image]

fwdrxyy, 2025-01-04:
As far as I know, Only Switch 3rd party controllers work. So for PS3, you have to get a OEM one.

GaryOderNichts, 2025-03-05:
Do you have a product link for this controller? Some 3rd party controllers do non-standard things and aren't supported.

FERFAIL-TXZ, 2025-04-07:
> Do you have a product link for this controller? Some 3rd party controllers do non-standard things and aren't supported.

Sorry for answering one month later. I don't exactly know what the product link is. Someone purchased this controller locally and gifted it to me. This is all I can show[image]