GaryOderNichts/Bloopair #167: 3rd party ds4 controller not working

andresitolamovie, 2025-02-09:
Hi! I haven't found this issue on the forum yet regarding latest updates (apologies if it has been already published). The thing is when I pair my 3rd party dualshock 4 controller it seems to connect to the console but when I press any key nothing happens. This happens when I use the latest version (1.0.2) but when I downgrade it to 5.0.3 it works normally. The thing is that I want to manually map the keys but koopair is only available on recent releases.

GaryOderNichts, 2025-03-05:
This has been fixed in https://github.com/GaryOderNichts/Bloopair/commit/333fb533d21670e359004726b15433daf1591163 and is available in the [latest nightly builds](https://nightly.link/GaryOderNichts/Bloopair/workflows/ci/main?preview).

rodri1421, 2025-03-07:
> This has been fixed in [333fb53](https://github.com/GaryOderNichts/Bloopair/commit/333fb533d21670e359004726b15433daf1591163) and is available in the [latest nightly builds](https://nightly.link/GaryOderNichts/Bloopair/workflows/ci/main?preview).

Works great!

rodri1421, 2025-03-09:
After a while the third party PS4 controller stops receiving inputs while the LED light remains on, I turn it off with the Playstation button and after 2 to 3 minutes it freezes again.

GaryOderNichts, 2025-03-09:
Which third party controller are you using exactly?

rodri1421, 2025-03-10:
> Which third party controller are you using exactly?

It's strange, because in version 0.5.3 I can use it without any problem...