GaryOderNichts/Bloopair #171: Ds3 third party inside koopair works but not outside

TedEkstrom, 2025-03-27:
Hi!
I have a interesting issue. If I'm inside thr Koopair app I'm able to pair my remote and it works great. But directly when I go out side and starts It up it indicates it's recognize the remote but it turns off right away. Some times it thinks the remote it's out of battery but some times it shows full charge but turns off anyway. Have any body some thoughts?

GaryOderNichts, 2025-04-07:
Odd, Koopair reads controller data like any other title. If it works in Koopair it should work outside of Koopair as well.
Can you upload a video of this behaviour?

TedEkstrom, 2025-04-07:
Yes i will do that. The thing is that I bought a real controler, a ds4, and tried that one. This controler works as I should.

TedEkstrom, 2025-04-09:
Hi again, here is the chip. I hope you are able to watch it: https://youtube.com/shorts/le4mz3iF-oQ?si=pK62E0EjtT1TIgsf

TedEkstrom, 2025-04-10:
No me neither. But could it be because you have more control over the controler inside the app?

GaryOderNichts, 2025-04-10:
Koopair uses the controller like any other title. That's what's so confusing about this.