GaryOderNichts/Bloopair #172: Use Gyro for Splatoon Battle Dojo

ok-andy, 2025-04-10:
Currently, because Bloopair emulates a Wiimote + Classic Controller, the hidden Splatoon 1 gyro mode in Battle Dojo does not work. It is impossible to pair another Wiimote to the controller to use the gyro sensor because the game assumes that the player wants to use the gyro built into the (virtual) Wiimote. The solution to this would be to either emulate Wiimote gyro OR make bloopair emulate a Wii U Pro Controller instead.

GaryOderNichts, 2025-04-10:
Bloopair can only emulate Wii U Pro Controllers. What makes you believe it's being detected as a classic controller?

ok-andy, 2025-04-11:
Nevermind, I just tried it again, gyro works in the Battle Dojo. I was doing something wrong on my end, sorry.

closed 2025-04-11 by ok-andy