GaryOderNichts/Bloopair #180: Switch 2 Pro Controller & GC NSO controller

BigBrainAFK, 2025-06-08:
Hello,

I wish I had the knowledge in C/C++ land to contribute addition of these myself.

Are these planned to be supported?

There is some tools that make them compatible with PC already (although seemingly a bit buggy for the GC NSO C stick and no Analog triggers)

[Here the HTML/JS](https://github.com/HandHeldLegend/handheldlegend.github.io/blob/master/procon2tool/index.html)

[Here the website running it](https://handheldlegend.github.io/procon2tool/)

Invictaz, 2025-06-08:
Bluetooth support is most likely BLE which the WiiU doesn't support.

andreiborisov, 2025-09-05:
What about wired connection?

Sjdhshdjdb, 2026-09-13:
Hallo wird daran gearbeitet

GaryOderNichts, 2026-10-03:
There are experimental Bloopair test builds available at https://github.com/GaryOderNichts/Bloopair/discussions/217 with support for BLE controllers.

GaryOderNichts, 2026-10-25:
[Version 1.1.0](https://github.com/GaryOderNichts/Bloopair/releases/tag/v1.1.0) now supports Switch 2 controllers.

closed 2026-10-25 by GaryOderNichts